zmkfirmware/zmk #422: Windows Dev Setup (Suggestions)

mcrosson, 2020-11-25:
I recently went to setup the native Windows development environment on Windows 10 version 20H2 and ran into troubles with the ```choco install dfu-util``` step. From what I was able to figure out with help from the Discord channel the ```dfu-util``` choco package is no longer maintained or install able via the tool.

The docker instructions, on the other hand, worked great and I've had no issues using the docker + vscode approach on an Intel i7 CPU *or* an Intel i5 CPU. The i5 is a little slower but not prohibitively slow. As long as a user has 4Gb of RAM they shouldn't run into any hassles using this approach on Windows.

Would it be prudent/advisable to update the documentation to use this approach by default for Windows users? If so, where in the docs should be updated? I may have some time in the coming weeks to submit the necessary documentation updates to default the docker + vscode approach for Windows users.

KingCoinless, 2020-11-25:
I'm inclined to agree with this approach. I had far too many issues associated with CMake on Windows which were completely eliminated when running Docker with VSCode. I've heard others having issues with Choco or DFU-Util as well. Unaware users on Windows will likely select the 'Windows' option over 'VSCode + Docker', and have to deal with a fidgety and arguably less stable process as a result. If developers don't want to dual-boot or use a Linux system, they should utilize VSCode + Docker to avoid this conundrum.

mcrosson, 2020-11-27:
Based on feedback in Discord and a previous PR, closing this issue as native builds on Windows should still be maintained.

If anything some better organization of the setup docs may be a better approach.

closed 2020-11-27 by mcrosson