zmkfirmware/zmk #432: Feature Request: USB polling rate adjustment feature

KingCoinless, 2020-11-27:
Offering support for USB polling rate adjustment would be a notable feature for users looking to use their ZMK-enabled devices in low-latency wired scenarios. For example, gamers and music producers rely on low input latency to achieve the most reactive and well-timed results possible. A simple and effective method would be to offer four standard tiered values of 125Hz, 250Hz, 500Hz, and 1Khz. These equate to 8ms, 4ms, 2ms, and 1ms average delay respectively. Users could then select one of these options and stage it as the default polling rate to utilize instantly upon USB connection. Specified key commands could be implemented as well for on-the-fly adjustment, but that is a secondary feature to this one.

joelspadin, 2020-12-18:
I believe this can already be adjusted with https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_USB_HID_POLL_INTERVAL_MS.html

KingCoinless, 2020-12-18:
@joelspadin I agree this approach (and also do not know why they chose 9 over 8ms). A default of 1ms (1Khz) polling for all devices when wired would be ideal. I mentioned the idea of offering alternatives as some people prefer the feeling of higher values (usually 500Hz), or find them more stable in certain applications (such as in FPS games like CSGO). Realistically, the 1kHz (1ms) default is certainly the best choice broadly-speaking.

joelspadin, 2020-12-18:
You should be able to set it in a .conf file. I don't currently have a good way to test that it's doing anything though.

joelspadin, 2020-12-18:
It was also mentioned on Discord that some platforms could have issues running very high polling rates. It would be good to get some data on how different devices (low and high end computers, phones, etc.) handle the different polling rates before we change the default.

KingCoinless, 2021-02-13:
As recommended by @joelspadin , I have tested this function in my Helix conf file using the following line:
`CONFIG_USB_HID_POLL_INTERVAL_MS=1`
In my raw build logs located [here](https://github.com/KingCoinless/zmk-nc/actions/runs/560695843), while I do not see it listed explicitly upon search for the left half, on solely the right, I get these lines:
[image]
I'm unsure if both sides are successfully using this configuration option or not. However, I will note that when testing latency on [Keyboard Scan Rate Tool](http://blog.seethis.link/scan-rate-estimator/), the registered key presses seem shorter than before. As this is subjective and subject to inaccuracy, I cannot say whether it is working or not. I think this is worth looking into, as well as adding a section to the documents for user reference once functionality is confirmed.

KingCoinless, 2021-02-13:
I received some feedback from @Nicell on the ZMK Discord server within the Development channel. He confirmed that the USB polling interval option does in fact work as intended for him, and that the log for the right side is simply due to it not communicating over USB in the first place. You can refer to that in this image:
[image]
Or, through this [link](https://discord.com/channels/719497620560543766/719544230497878116/810050995320389642). I will leave this Issue open for future discussion as Nicell advised.