zmkfirmware/zmk #445: Sticky key in a sticky layer issues, wrong output on the 2nd consecutive sticky key, then sticky key is ignored.

yannickjmt, 2020-11-29:
key OSL: &sl OSL in base layer
key s: &sk LSHFT in the layer OSL, &kp S in base layer
key k: &trans in the layer OSL, &kp K in base layer

scenario1.log
sequence: [OSL, s, k] short pause [OSL, s, k]
on screen: "Ksk"

scenario2.log
After multiple tries (a dozen?), the shift seems to stopped working completely.
Same sequence as scenario 1, but "ksk" output

[scenario1.log](https://github.com/zmkfirmware/zmk/files/5612252/scenario1.log)
[scenario2.log](https://github.com/zmkfirmware/zmk/files/5612253/scenario2.log)
[Kyria_keymap.txt](https://github.com/zmkfirmware/zmk/files/5612254/Kyria_keymap.txt)

okke-formsma, 2020-11-29:
I'm taking a look

okke-formsma, 2020-11-29:
The sticky key was not cleaned up properly in some cases. Bugfix PR incoming.

closed 2020-11-30 by petejohanson