zmkfirmware/zmk #448: Sticky keys issue: stuck modifier when pressing the same key as the one assigned to the sticky key.

yannickjmt, 2020-11-30:
Sorry for the bad title, it's late here and could not find a clear one liner 😅
tap &sl OSL, tap &sk LSHFT, tap &kp S gets a stuck shift modifier (&sk LSHFT and &kp S are the same physical key)

When the key is not repeated but another &sk key, the key seems to be ignored:
tap &sl OSL, tap &sk LCTRL, tap &kp S doesn't output the expected CTRL(S)

okke-formsma, 2020-11-30:
Looking into it.

okke-formsma, 2020-11-30:
The stuck key is fixed in https://github.com/zmkfirmware/zmk/pull/449.

your entire issue is not fixed though; in your case, the LSHFT will be released before S is pressed because it re-uses the same key position.

okke-formsma, 2020-12-06:
@yannickjmt can you verify that this issue is resolved in the current ZMK?

yannickjmt, 2020-12-06:
The sticky mod issue is solved, thanks!
But there is kind of a weird behaviour. I put sticky modifiers on ASDF keys. In order to do a ctrl A for example, I need to double-tap the sticky ctrl key then hit A. But a single tap works for any other key than ASDF.
tap &sl OSL, tap &sk LCTRL, tap &kp A doesn't do anything but 
tap &sl OSL, tap &sk LCTRL, tap &sk LCTRL, tap &kp A works.

closed 2020-12-06 by yannickjmt