zmkfirmware/zmk #453: Towards a modular ZMK

innovaker, 2020-12-01:
Fellow developers, thank you for your patience.  The modular proof-of-concept (PoC) is ready for evaluation.

# Purpose

This is a (fully functional) **proof of concept**.  Its purpose is to:
- enable everyone to explore what a modular ZMK would entail.
- act as a point of reference for discussions about modularity.
- be the basis for properly evaluating a modular ZMK, including:
  - technical design
  - end-user experiences
  - developer workflows and experiences
  - maintenance
  - best practices
- be the basis for moving forward if we choose to go down this road (it's a high fidelity prototype).

# What's "a modular ZMK"?

- Modules:
  - can contain boards, shields, C (code), Kconfig, dts, soc, or arch.
  - can live anywhere in their own git repository (web or filesystems).
  - can be owned, licensed and/or maintained by anyone (distributed development).
  - are facilitated by Zephyr's `west` tool.
- We distill the ZMK repository down to core components only.
  - Non-core components (boards/shields,etc.) can be moved into external (optional) modules.
- We supersede `zmk-config` with a simple user "application".
  - This "application" module/repository is owned by the end-user.
  - It's specific to their own build.
  - It sits in the driver's seat.
    - It describes:
      - Which modules their application/build depends on (including boards, shields or extra features).
      - Any configuration or code specific to their build.
    - It's square one for the build process.
  - It can be as simple or complex as the user desires.
- ZMK effectively becomes a core module/library/SDK.

# Motivations

I'm suggesting we consider a modular approach for ZMK because of several pressing issues, including:
- Board and Shields
  - As ZMK grows in popularity we're seeing the number of board and shield PRs increase.
  - We're facing pressure around which contributions we should accept into the core.
  - Any time core contributors spend on boards/shields is time taken from core development, slowing the project down.
  - The more boards and shields we accept, the more bloated ZMK becomes and the more maintenance baggage we take on.
  - The more we take on, the more impractical or unaffordable it will become to test them as part of ZMK's CI.
  - Our current approach is great for popular boards/shields, but does little for the smaller players.  We risk artificially limiting discoverability.
  - We maintain pretty high standards for PRs, which isn't everyone's cup of tea.
  - What happens when contributions become abandoned?  How do we deprecate them?
- Distributed Extensions
  - Currently there are only these options for adding functionality:
    - Add it to your `zmk-config`.
    - Create a branch/fork of ZMK.
    - Submit a PR to add it to the core (which is an understandably slow process).
  - These approaches get the job done, but none of them are particularly great for sharing with others.
  - It's not easy for developers and testers who often live on the cutting edge from multiple sources.
- Flexibility
  - Modularity would let us explore and adopt a range of approaches for the future, such as:
    - Hardware developer owned & maintained modules (boards/shields).
    - Author owned & maintained modules.
    - Fully distributed boards/shields (each in their own module).
    - Community maintained modules/repositories.
  - Users could leverage other Zephyr modules such as the [Nordic nRF Connect SDK](https://github.com/nrfconnect/sdk-nrf) without licensing issues for ZMK itself.
  - Users could have more control over the build.

# First steps
- I've built a demo which uses an in-development version of the `tidbit` shield.  Thanks let us use it Kemonine!
- The shield itself lives in its own [repository](https://github.com/innovaker2/zmk-tidbit)/module.

## Local (developers with existing environment setup)
This includes developers using Windows, macOS, Linux or WSL2.
```
[... 69 lines cut ...]
    - This will let you see the `zmk` folder and give you access to the other VS Code workspace benefits.
- Open a terminal for `app` and run: `west build`

Note, I've configured it to mount `zephyr` and its dependencies into (shared) Docker volumes so that it doesn't take an eternity to run `west update`.  These volumes are shared across containers and not junked when you delete a container.  `zmk` is currently stored within the container itself (so it will be lost/re-cloned if you rebuild the container) but it too can be mounted as a volume by a savvy user (I've left it commented out).

### Codespaces + nested branch

I've not personally tested it on Codespaces yet but my cursory understanding is that it's built around the mono-repository paradigm and doesn't support untracked flat structure workspaces???  In anticipation of that, I've' been exploring a nested version of `zmk-app`/`my-zmk-keyboard`.  I'm not keen on it personally but I made it as an experiment.

The branch names are `nested-v1` if you want to try it.

# Next steps
- Edit: `CMakeList.txt`
- Drop in: your own overlay/keymap?
- Rebuild with: `west build -p` (pristine build)
- Explore: `west.yml`
- Try: making your own module?

# Going forward

- Whilst exciting, modular development is a complex topic which merits careful consideration and debate.
- It would entail a major paradigm shift to our current approach.  We shouldn't get carried away and rush into it.
- We should use this PoC to explore the pros and cons together.
- I encourage you to actively try developing with it, particularly to get a feel for its effects on workflow.
- We need to find the pain points and potential blockers.

innovaker, 2020-12-01:
For those wishing to dig deeper:
- the ZMK core changes can be found on the [modular-v1](https://github.com/innovaker/zmk/tree/modular-v1) branch, they're self documenting.
- the upstream of [my-zmk-keyboard](https://github.com/innovaker/my-zmk-keyboard) is [zmk-app](https://github.com/innovaker/zmk-app)
- the upstream of [zmk-tidbit](https://github.com/innovaker2/zmk-tidbit) is [zmk-module](https://github.com/innovaker/zmk-module)
- the general idea behind the upstreaming is that we clone copies of `zmk-app` or `zmk-module` to create new user applications or modules.  Infrequent upstream updates can then be merged in.
  - for modules, the developers/maintainers can do this and we could provide a convenience GitHub workflow to notify them
  - for applications, we might achieve this with:
    - scripts for the initial clone (similar to `setup.sh`)
    - React based system on the website (using API calls to GitHub)
    - scripts embedded into `zmk-app` itself (for updating / merging)
  - For users who are incapable of merging, we can just suggest cloning a fresh application and copying their key configuration files or changes across.

innovaker, 2020-12-01:
For those out the loop, the general ideas for discoverability of modules so far has been:
- Maintaining an index on the website.
  - Module authors can PR a metadata file into the core.
  - Compilation of the metadata can be driven by webpack/netlify.
  - React/Docusaurus can be used to provide a explorable interface with search, categories and generate appropriate `west.yml` entries.
  - We could perhaps tie the front-end into the GitHub API as a stretch goal.
  - I appreciate this smells like a package repository.
- Encouraging a naming convention to make it easier to find ZMK modules via GitHub's search.

innovaker, 2020-12-15:
I've released v2 of the modular PoC, which is (re-)based onto zmk's `main` as of today (f3502dbcb5f017e0b62b9cf09cb0e81c26ce4589 which includes Zephyr v2.4.0).

I've updated the previous posts accordingly.  If you've already tried v1 with Docker, you may have to remove the old containers before trying v2.

I have no intentions of making another release in the foreseeable future as it's quite an intensive process.

Nicell, 2021-01-07:
While testing this on Windows I found an issue with finding module root paths causing mixed styles of path separators, which seems to be fixed in https://github.com/zephyrproject-rtos/zephyr/pull/29243, but it isn't in a stable release of Zephyr yet.

For now I've manually added this fix to my local development workspace, but we should be aware of it when getting closer to a full implementation of this.

petejohanson, 2021-01-30:
Some initial thoughts:

* While making ZMK a "core library" is generally a great move, 95% of users are unlikely to ever want to have a custom`main` entrypoint, versus having their custom code just be in their module w/ init callbacks, etc. Thoughts on having our core library still add a `main` unless a Kconfig option is set to disable that? In that scenario it's disabled, we would still export an entrypoint that the user `main` could invoke.
* Doing that possibly removes need for a CMakeLists.txt by default in the "user application".

Upstream modules:

I think templates like this are really important, ideally they can be as minimalist as possible while still offering the full features. Something I don't love about the current templates for the user config repos is them being external to the core repo, so changes that need coordination are harder.

I would propose those templates actually be in the main repo, and creating new downstream copies can pull from the subdirectory of the core

innovaker, 2021-02-04:
For the benefit of casual observers, there were some further discussions on Discord:
- https://discord.com/channels/719497620560543766/719544230497878116/806393254593429575
- https://discord.com/channels/719497620560543766/719544230497878116/806483474541707304

innovaker, 2021-02-04:
> While making ZMK a "core library" is generally a great move, 95% of users are unlikely to ever want to have a custommain entrypoint, versus having their custom code just be in their module w/ init [...]

This makes sense and I'm fine with it.  I also tested the principle last night.  It's serves the majority but facilitates the minority.

innovaker, 2021-02-04:
> Doing that possibly removes need for a CMakeLists.txt by default in the "user application".

I doubt the bootstrapping code can be removed from the application but I'm not a cmake expert and welcome alternatives.

innovaker, 2021-02-04:
> I would propose those templates actually be in the main repo, and creating new downstream copies can pull from the subdirectory of the core.

I wasn't  aware that pulling from a subdirectory of a git repository was possible (unless you're talking about partial clone?)?   That's partly why I went for an upstream module approach in the PoC.

schengnz, 2021-03-04:
Is there any possibility of decoupling the key processing engine as a separate module, independent of LED, key-scanning, sensors, display, power management and BLE/USB HID support? That way a key processing engine can be swapped in and out as out.

Anutrix, 2023-10-28:
It's been almost 3 years. Has there been no updates or was this just not updated?

caksoylar, 2023-10-29:
@Anutrix This is a very general framework, and it is partially implemented already in how currently user config repos work (where `zmk` itself is used as a Zephyr module). Maybe you should note what exactly are you looking to accomplish and what issue are you running into with the current limitations.

Anutrix, 2023-10-29:
I am looking for support on ESP32-S3 which is supported officially by Zephyr Project(https://docs.zephyrproject.org/latest/boards/xtensa/esp32s3_devkitm/doc/index.html; https://github.com/zephyrproject-rtos/zephyr/commits/main/boards/xtensa/esp32s3_devkitm has had most features added by now.) but not ZMK.

It seemed that Modular ZMK would make adding support for it easier before I can deep type into this repo.

caksoylar, 2023-10-29:
You can directly use boards from Zephyr in ZMK but you'll typically need to enable certain options, see some examples like for Adafruit KB2040 ([overlay](https://github.com/zmkfirmware/zmk/blob/main/app/boards/adafruit_kb2040.overlay) and [conf](https://github.com/zmkfirmware/zmk/blob/main/app/boards/adafruit_kb2040.conf)). You could even do these overrides in a config repo, under `config/boards/*.overlay` etc.