zmkfirmware/zmk #460: settings_reset shield only reset Bluetooth settings

Nicell, 2020-12-02:
Currently the settings_reset shield only resets Bluetooth settings, we should probably expand this to clear out the entire NVS storage, as many more parts are starting to incorporate settings.

jmding8, 2021-06-07:
Would it be possible to trigger this reset behavior from a keycode, instead of having to flash it?

My MBP occasionally disconnects from my keyboard, where the only way to restore the connection is to go through the process of flashing the settings_reset shield, which is quite a hassle

Lenbok, 2022-06-10:
I get this too on linux where sometimes the only way to get a connection re-established is to flash the settings reset. It would be immensely helpful if there were a way to bind this to a key instead of having to reflash twice.

closed 2025-07-31 by nmunnich

joelspadin, 2025-07-31:
Done in 1dfcfc7d3f074e538900a6df684cc99e92c60a21