zmkfirmware/zmk #490: Feature Request: Double reset/bootloader for split keyboards

jmding8, 2020-12-07:
Currently for split keyboards using Bluetooth for inter-half communications, &reset only triggers the central device to reset, while the peripheral device does not. This causes the two halves to disconnect, requiring the user to do a hardware reset on the peripheral device to re-connect the halves. Similarly, &bootloader causes only the central device to enter bootloader mode, requiring the user to trigger bootloader mode on the peripheral device via a reset-pin double-tap. It would be nice if there were &reset_peripheral, &bootloader_peripheral, &reset_all and &bootloader_all behaviors available

Eyenseo, 2021-01-17:
Desperately needed - It would also be nice if `reset` is applied on each half - or take an index to say on which piece of a multi-piece keyboard.

petejohanson, 2021-01-17:
> Desperately needed - It would also be nice if `reset` is applied on each half - or take an index to say on which piece of a multi-piece keyboard.

Check out #547 which will be fixing this.

caksoylar, 2022-04-03:
Closing since #547 is merged.

closed 2022-04-03 by caksoylar