zmkfirmware/zmk #491: Multiple typing RGB_TOG results in zmk freeze

HendrikRoth, 2020-12-08:
Multiple times typing RGB_TOG results in zmk freeze, no control is reacting anymore.
keymap: https://github.com/ebastler/zmk/blob/sabrina75-rev2/app/boards/arm/sabrina75_rev2/sabrina75_rev2.keymap

Nicell, 2020-12-09:
@HendrikRoth I'm having a hard time recreating this issue. Are you working on the latest of `main`? How quickly are you pressing consecutively? Are you running into this issue when wired, wireless, or both?

ebastler, 2020-12-09:
Since I'm the one who made the firmware, I'll join in. When I created the firmware I was on the latest main branch, by now it's lagging 34 commits behind. 

The board has a power toggle MOSFET for the underglow, so RGB_TOG toggles both the ws2812 data and power line. 

I only ran into it while wireless, but I never had any switches connected to the board, so I didn't try too much apart from a quick check if all features work before shipping the PCB to Hendrik.

For me it died when I was bridging the rgb toggle with tweezers, and accidebtally switched on/off quickly due to bad contact between tweezers an pads, but Hendrik has switches in it by now.

Nicell, 2020-12-12:
@ebastler I have a similar set up on my nice!60 board (MOSFET on underglow which toggles with RGB_TOG), and I can't recreate this issue no matter how fast I press RGB_TOG over and over.

Maybe logging during the even could show us what exactly is breaking. (Is the timer not happy about being turned on and off so fast? is there some weird SPI issue?)

@HendrikRoth could you try logging or possibly explain more closely what you're doing on the board to make this happen? I assume it's a bit different with switches in.

mcrosson, 2020-12-15:
I just tried this during some tests with a nice!nano and tidbit and can *not* re-create.