zmkfirmware/zmk #497: ZMK Sponsorship?

petejohanson, 2020-12-10:
Several times, the topic of "I want to contribute back to ZMK efforts financially" has come up, and in the past I have responded with:, "We don't currently accept sponsorship; ZMK is a community effort, and it's not clear who would receive any $ from sponsorship".

That being said, there are some subscriptions/recurring costs it would be potentially nice to have as an organization:

* DNS Name Registration
* GitHub Pro for the organization for better controls/management features.
* Docker Pro account for container scanning, if we plan to encourage users to use our containers to buil/develop.
* Discord Nitro? Do we care?

In addition to that, $ could be put to:

* Purchasing stickers once we have a logo/mascot.
* Purchasing hardware for contributors to test/develop against
   * What hardware? Dev Kits? Keyboards? Something else?
   * For whom?

That then leaves the last, I think trickiest/hardest thing, which is:

* Do we at all pay any individuals directly?
* Should we use some $ to pay a designer so we get the mascot/logo done sooner?
  * This sets up an artificial difference between OSS coders and OSS designers. Not sure I like setting that precedent.

Note & Full Disclosure: I personally do not in any way need or want direct compensation for my general involvement in ZMK. The only time I might even consider direct payment is some theoretical scenario where a company wants to contract directly with me to pay me to work on a specific feature they want I wouldn't otherwise prioritize. I have no current plans, nor any direct requests from anyone to do such a thing, but that's the only scenario I might personal decide to accept $ for ZMK work.

Thoughts? Should we develop a plan to start accepting sponsorship/$ contributions for any of the above?

mcrosson, 2020-12-10:
In the past I've leveraged donations to OSS projects to...

- setup community members with new hardware via give aways
- hardware for regular contributors with a stipulation to 'pay it forward' if/when they are no longer active contributors
- cover regular hosting costs
- pay for stickers that can be given away
- pay a designer (even a non-oss designer) for a logo or other artwork that's then released under a CC-BY license to the public

Some other ideas that I can think of

- Pay a board maker / fab to do a small batch of mcu boards to increase availability
- Partner/pay a supplier to make parts more generally available that are commonly "lacking" for supply
- Pay it forward to the zephyr team if they are accepting donations
- Bug and feature bounties

petejohanson, 2020-12-10:
> In the past I've leveraged donations to OSS projects to...

Thanks for the ideas!
 
> * setup community members with new hardware via give aways

What kind of hardware? Like, community giveaways of BLE keyboards? Not sure that's top of my list in terms of priorities. Seems like that's marketing, more than fostering contribution.

> * hardware for regular contributors with a stipulation to 'pay it forward' if/when they are no longer active contributors

Yeah, this is one I had considered. In particular, often folks right now will have nice!nano or other board, that might be good for use, but doesn't allow as easy debugging as something like a dev kit. Could sponsor Nordic nrf52840 DKs, or MakerDiary nRF82840 MDK devices for folks once they have contributed "X".

> * cover regular hosting costs

Yeah, this one is clearest cut to me, e.g. GH Pro, Docker Pro, DNS registration.

> * pay for stickers that can be given away

This is high on my list too. Send a pack of stickers to everyone on their first PR merge for ZMK.

> * pay a designer (even a non-oss designer) for a logo or other artwork that's then released under a CC-BY license to the public

Yeah, this may be what we need to do. See my comment about reservations about dev vs designers and pay though.

> 
> Some other ideas that I can think of
> 
> * Pay a board maker / fab to do a small batch of mcu boards to increase availability

Interesting idea. You mean wireless pro-micro boards like the n!n? That is an idea, but sets us up to *compete* with other vendors. QMK has gone that route w/ the proton-c... Which then some vendors sell. If we were to do that, I'd honestly rather get some a small onboard macropad produced for folks that want a board to play on, which is less "compete with other vendords", a bit.

> * Partner/pay a supplier to make parts more generally available that are commonly "lacking" for supply

What sort of parts?

> * Pay it forward to the zephyr team if they are accepting donations

Zephyr has sponsorship model already, yes. Most sponsors are much bigger than us, with deeper pockets. Maybe a future time?

> * Bug and feature bounties

This is an interesting way to possibly get $ to contributors that is a bit more specific/tangible. Not sure about what sort of incentives this creates. Would need to ponder that.

petejohanson, 2020-12-10:
Note: I think the only way I would be happy with putting up $ to pay a designer is if we *also* set up bug/feature bounties for developers as well. That would be a way to "level the field" for "how do you get paid to contribute to ZMK".

mcrosson, 2020-12-10:
> > * setup community members with new hardware via give aways
> 
> What kind of hardware? Like, community giveaways of BLE keyboards? Not sure that's top of my list in terms of priorities. Seems like that's marketing, more than fostering contribution.

This would definitely be marketing if it's "no strings attached" kind of give aways.

> Yeah, this is one I had considered. In particular, often folks right now will have nice!nano or other board, that might be good for use, but doesn't allow as easy debugging as something like a dev [...]

Exactly what I was thinking.

> > * pay a designer (even a non-oss designer) for a logo or other artwork that's then released under a CC-BY license to the public
> 
> Yeah, this may be what we need to do. See my comment about reservations about dev vs designers and pay though.

The trouble is few designers work for free or on a pro bono basis. Programmers and EE types are far more willing to contribute "for free". There is very much an ethos difference between the two.

> Interesting idea. You mean wireless pro-micro boards like the n!n? That is an idea, but sets us up to _compete_ with other vendors. QMK has gone that route w/ the proton-c... Which then some vendors [...]

You could put it to a community vote for which vendor gets boot strap cash for a run of boards. You could also have it be a business deal of sorts where the cash gets paid back from board sales. You're bootstrapping the supply chain more than competing at that point.

> > * Partner/pay a supplier to make parts more generally available that are commonly "lacking" for supply
> 
> What sort of parts?

Anything that's a pain to obtain. Right now that's MCU boards but it could also be switches or anything else where the supply chain at the time is lacking.

> > * Pay it forward to the zephyr team if they are accepting donations
> 
> Zephyr has sponsorship model already, yes. Most sponsors are much bigger than us, with deeper pockets. Maybe a future time?

If they have a "low cost" teir, just as valid as a bigger donation. We may be "small" but it's still a good option for payihng things forward IMO.

> > * Bug and feature bounties
> 
> This is an interesting way to possibly get $ to contributors that is a bit more specific/tangible. Not sure about what sort of incentives this creates. Would need to ponder that.

I have a feeling it'll look a lot like some of the other bounty programs but adjusted based on the actual needs zmk has: code for cash. If left too open ended, it can incentivize drive by contributions for cash and could alienate core devs if they aren't wholly on board from the outset. I imagine for a smaller group of devs like zmk has it'd need to be a very specific and very targeted bounty that is related to something that can't seem to gain enough traction to become a priority relative to other work that needs to be done day to day.

petejohanson, 2020-12-11:
Digging a bit, if we think there are enough of these kinds of efforts the community feels are worthy, we could chose to get started with something like Open Source Collective, to start collecting some sponsorship now, then continue evaluating how and what we should apply things to.

It seems doing this at a minimum for:

* Hosting/service costs
* Stickers

Is solid, and we could evaluate other efforts from there.

mcrosson, 2020-12-11:
> Digging a bit, if we think there are enough of these kinds of efforts the community feels are worthy, we could chose to get started with something like Open Source Collective, to start collecting [...]

I can say positive things about OpenCollective. I've used them in the past and they are solid for an option.

Nicell, 2020-12-11:
I think there's a bunch of really great ideas here. Can't forget about #38! Although that one may take a bigger more targeted campaign to reach. I think it makes sense to start an Open Collective. We can flesh out how the funds will be used as we go, but for now we can start collecting.

mcrosson, 2020-12-11:
> 
> 
> I think there's a bunch of really great ideas here. Can't forget about #38! Although that one may take a bigger more targeted campaign to reach. I think it makes sense to start an Open Collective. We can flesh out how the funds will be used as we go, but for now we can start collecting.

OpenCollective lets you define budgets as well as campaign/milestone goals. It could be setup to show the hosting costs (they are BIG on radical transparency, including who gets paid and why) as well as a longer term stretch goal for a proper usb id.

petejohanson, 2020-12-11:
> OpenCollective lets you define budgets as well as campaign/milestone goals. It could be setup to show the hosting costs (they are BIG on radical transparency, including who gets paid and why) as [...]

Yeah, I saw the transparency stuff for all payments from the given project/collective, etc. I really like that.

Especially, since as the project founder, I *really* want to be transparent about not setting up the org for sponsorship so I get paid personally.

innovaker, 2020-12-11:
I'm in favor of initially covering running costs with donations.  I also feel we should take it slow and not jump into anything too hastily.

Regarding individual sponsorship of developers, I'm not completely opposed to it, however I've strong reservations around how it can potentially influence design decisions and priorities.  Given that we're not a business, I feel that design decisions should come down to the best ideas and rationales, rather than descending into a "deepest pockets wins" approach.  I can see potential for conflicts of interest that wouldn't benefit the project in the longer term.

Development kits for core developers would also be worth looking at, particularly if users start requesting support for new chips.

petejohanson, 2020-12-11:
> I'm in favor of initially covering running costs with donations. I also feel we should take it slow and not jump into anything too hastily.

Fully agreed. Phased approach.
> 
> Regarding individual sponsorship of developers, I'm not completely opposed to it, however I've strong reservations around how it can potentially influence design decisions and priorities. Given that we're not a business, I feel that design decisions should come down to the best ideas and rationales, rather than descending into a "deepest pockets wins" approach (or worse still: the cheapest option). I can see potential for conflicts of interest that wouldn't benefit the project in the longer term.

I have similar reservations about this. The main reason I don't want to dismiss it outright is to make sure I'm not using my financial privilege (I don't happen to need $ to have time to contribute to ZMK) to leave a blind spot to something that other contributors might really need. Still a potentially slippery slope.

> 
> Development kits for core developers would also be worth looking at, particularly if users start requesting support for new chips.

Yeah, that was something I was thinking too, e.g. the new nRF5340 stuff.

petejohanson, 2020-12-11:
I've created an OpenCollective collective for ZMK: https://opencollective.com/zmkfirmware

I will work on getting it setup w/ GH Sponsors stuff. From there, if/when we get some contributions, we can select the specific subscriptions, etc to start using.

bennovack, 2021-01-04:
I just wanted to throw in a comment about design vs development: Unpaid dev work has long been part of the OSS ethos and, if nothing else, it often is a key part of interviews and portfolios. But developers as a rule have relatively few frustrations with folks approaching them and expecting or asking for free dev work. Most of the designers I know are annoyed with frequent requests for free work - often for for-profit businesses - and have less trouble building a portfolio (by definition most of their work is public-facing or at least less sensitive than code.) Adding to this, to overgeneralize, most developers are much better-paid than most designers, especially below senior levels. The professions have different relationships to free labor and therefore I don't think it's unreasonable, at all, to actively avoid paying devs for their time while still carving off a slice of donation revenue for marketing/logo design work.

petejohanson, 2021-01-04:
> I just wanted to throw in a comment about design vs development: Unpaid dev work has long been part of the OSS ethos and, if nothing else, it often is a key part of interviews and portfolios. But [...]

@bennovack This is a fair assessment of things. I definitely agree that the sentiment about "free labor" in design is very different. I mean, @forexposuretxt on Twitter highlights that problem regularly.

I am leaning toward using some of our collected funds for a fiverr or freelancer.com effort to get our logo work done.

maxderc00le, 2021-03-15:
Hello there,
I would add that we could have a pool of money(if for example individuals or companies want to donate) and the community can vote on different issues that have a certain need but are open for a longer time(or depending on the time there may be different limits on the amount) -> probably hard to solve. This would give people an incentive to work on the really hard stuff. I would of course give the contributor(s) the option of either donating to charitable effort/leave the money or their part in the pool. This could be money that is unused after being able to pay all bills for the next years or given just for this cause.

closed 2025-12-26 by nmunnich