zmkfirmware/zmk #544: tap-only combos

okke-formsma, 2020-12-26:
from makenova on [discord](https://discord.com/channels/719497620560543766/784523018050469939/791743362964258886):

> Right now I use home-row-mods(hrm) but I also have combos on the home row.
> My hrm are in the form of SCAG(asdf) and I also have a combo on DF.
> If I want to hold alt + gui right now I have to hold one then the other or else the combo triggers if I tap them together.
> With what I understand of the tap-only combos, there will be another constant defined which is > combo_term(time that a combo has to be held to trigger).
> If the combo is held longer than the new constant, then the combo is not triggered but the hrm are.

okke-formsma, 2021-01-04:
oops, sorry for the assignment. I thought I had a PR open

urob, 2022-06-10:
Seconding that this would be useful. I am using tap-only combos from the early combos branch for QMK and have found it very useful for precisely the reasons stated by OP. Without it, chording HRMs that also have a combo defined becomes very cumbersome, which is unfortunate, because by design HRMs involve the most ergonomic key positions.

anderso, 2023-06-26:
This is what keeps me from using zmk. Are there any workarounds?

urob, 2023-06-26:
> This is what keeps me from using zmk. Are there any workarounds?

### Workaround

Binding the combos to hold-taps with a hold-action equal to the corresponding combination of mods works. See [example](https://github.com/urob/zmk-config/blob/c21715bd1dc08738c1ff43ff63f1e611fa53f1de/config/combos.dtsi#L17-L21).

The downsides relative to a native solutions are:

- It's more cumbersome, especially when the combo is for behaviors other than `kp` and `lt`. Wrapping the hold-tap config into a macro as in the example above can alleviate the extra burden.
- Dynamically adding and removing mods to a chord doesn't work. Say, you start holding index + middle, which trigger, say, `LCTRL` + `LSHFT`. Now adding the ring finger won't work.

On the upside, going this route allows you to fine-tune when the combo triggers with all the hold-tap config options (`global-quick-tap`, `positional-hold-tap` etc) at your disposal... 

### Thoughts about a native solution

I see two approaches. 

1. Replicate the hold-tap logic and only trigger combos that satisfy certain criteria. One will need to decide what trigger-logic to adopt and/or whether one would expose any options? Maybe it's also possible to build on the existing hold-tap implementation internally?
2. Instead of adding a trigger-logic to the combo itself, maybe one could add a `trigger-on-hold-tap-holds` flag. If set to True, the combo would recognize that a candidate is a hold-tap, wait until it is resolved, and only trigger if it resolved as tap.

The benefit of the second option is that it would piggyback on the hold-tap behavior to see what it resolves to. As a freeby, one would get *all* the configuration options already existing for hold-taps. Implementation wise, it would probably require patching both the hold-tap and combo behaviors.