zmkfirmware/zmk #554: per-key led feature

Retic1337, 2020-12-29:
creating this to start work on the per key led lighting.

joelspadin, 2020-12-29:
I think the first thing to do is to get a list of use cases for per key lighting. Here are a few I can think of:

1. Set different solid colors per key (e.g. mods one color and everything else a second color).
2. Dim/disable lighting for keys that have no function in the current layer.
3. Status LEDs (caps lock, num lock)
4. Static animations (e.g. color wave across keyboard)
5. Dynamic animations (e.g. ripples from pressed keys)

Other considerations:

It would be nice to have one system that manages all LEDs and then have mappings on top of that for per key LEDs, underglow LEDs, and standalone status LEDs. Having separate systems for each of those would likely lead to inconsistencies and make it difficult to do things like creating an animation that syncs between key and underglow LEDs.

Just like with the existing kscan and composite kscan drivers, we should support multiple LED drivers (e.g. direct GPIO, daisy chained ws2818, Lumissil matrix driver) and combinations of them controlling different subsets of LEDs.

Storing colors as red, green, and blue bytes is probably the format that works best across all LED drivers. We can still convert to/from HSV as needed for things like adjusting the underglow color. To support single color LEDs with the same system, the driver can simply read the red channel and adjust a PWM or use an on/off threshold.

We need to somehow support wireless splits. Central could send per LED color changes to the peripheral, but that probably won't perform well with fast animations. The peripheral likely needs to run the same LED control code as the central and just get high level commands like hue/brightness changes from central.

Retic1337, 2021-01-04:
for #1 and #2 we will need to import the keymap into the driver to determine colors.
#3 will require reading status of caps, num, etc..
#4 i think would have to be a map of the leds position x and y relative to their placement on the pcb. so esc on a tkl would be 0,255 and right arrow would be 255,0. 
#5 would be easy once #4 is done.
Currently have a background color working with per key color changing based upon keypress. Where are we going to want to be creating led maps and defining colors etc, board.keymap? or in the device-tree?

joelspadin, 2021-01-04:
> for #1 and #2 we will need to import the keymap into the driver to determine colors.
> #3 will require reading status of caps, num, etc..

I think we should structure things so that the low-level LED drivers don't have any knowledge of the keymaps. Instead, something built on top of that (not sure if it would also be considered a driver or just application code) should be deciding what color each LED should be and sending commands to the LED driver(s). It is this higher level code that would need to know about the keymap and lock statuses.

One idea I had for this was to have a base LED map/animation that defines the color of each LED and then stack "transforms" on top of that which can optionally modify the colors before they are sent to the LED driver(s). For example, a status indicator could be a transform that overrides the color of one specific key based on the lock status. Dimming keys based on layer could be another transform that reads the keymap and overrides LEDs accordingly.

Pseudocode for how this might work:

```python
# User-configurable part
def base_animation(colors):
    for n in range(len(colors)):
        colors[n] = LED_MAP[n]

def caps_transform(colors):
    if is_caps_on:
        colors[CAPS_INDEX] = (0, 255, 0)

def layer_highlight_transform(colors):
    for n in range(len(colors))
        if not current_layer_has_function(n):
            colors[n] = (0, 0, 0)

TRANSFORMS = [layer_highlight_transform, caps_transform]

# This part happens the same way no matter how things are configured
colors = [(0, 0, 0)] * LED_COUNT

def update_colors():
    base_animation(colors)
 
    for transform in TRANSFORMS:
        transform(colors)

    send_to_driver(colors)
```

Both the base animation and transforms would also need a way to trigger a color update (e.g. periodic for an animation, on layer change for layer-based dimming, etc.).

> Where are we going to want to be creating led maps and defining colors etc, board.keymap? or in the device-tree?

The organization that makes the most sense to me is to define all the driver and hardware configuration in the board DTS/shield overlay and then put the LED maps and any other user configuration (e.g. enable/disable dimming keys based on layer, etc.) in the keymap.

KingCoinless, 2021-01-06:
No notes on how to write this yet, but I do have a recommendation for a specific kind of reactive per-key effect. On my Kemove 64/66 keyboard, there is a lighting mode which, when pressing any key, reacts on that key in a random color and disappears. Fn + Up and Down Arrow cause the brightness levels to raise or lower. Fn + Left and Right Arrow cause the effect's dissipation rate to speed up or slow down. It might be interesting to offer something like this in ZMK, or perhaps even a lighting mode which reacts around a specific color (for example, blue, or tints of blue). The color (or tints of) could be adjusted through the 'RGB_HUE' and 'RGB_HUD' key commands like normal. We could give the light effects a dissipation rate range of maybe 200ms - 2s, with nine steps of 200ms in between (200 -> 400 - > 600, etc). It's a very pretty effect, and one that I think would be great to include here.

ice9js, 2021-07-09:
I've been thinking about the kind of data structures that make sense to support this.  
Specifically how @joelspadin already [pointed out](https://github.com/zmkfirmware/zmk/issues/554#issuecomment-752262672), it'd be nice to have the animation 'engine' support different LED drivers and combinations of them. I'd even expand on those guidelines:

- One IC should be able to support multiple logic groups of the LEDs as well. Like, per-key-rgb AND underglow to give an extreme example.
- Make it possible to have different LEDs follow different animations even if the behavior is the same. One might have a different effect on the bottom and on the top, or define multiple light strips each responsible for its own thing. Effectively, fine grained control.
- Eventually all of this should be reconfigurable on the fly without having to flash the board.
- _It might be interesting to expose an API for other parts of the firmware to control the animation/parameters for a given area. Like, using your underglow LEDs as a charging indicator or supporting power on/off animations.

**Let's assume we'll be using Zephyr's led_strip API as a common starting point.**  
A single GPIO-driven LED can be represented as a strip with length of 1, a driver would be simple enough. Similarly, LED matrices can be transposed into a 1d array and vice versa.  
Thus, we will need a `pixels[]` array:

[image]

This would be a 1D array containing the values for **all** LEDs on the board.  
In the example above, LEDs `0...m-1` are driven by one IC while `m...n` are driven by another. Organizing it this way allows to pass the appropriate pointers straight to `led_strip.update_rgb()` saving memory and processing overhead.

That was the easy part. Now, let's add a second `pixel_meta[0;n]` array in parallel. This array would contain all the necessary settings for each pixel. For the purpose of illustration, I'm going to portray it as an array of structs. However it could just be multiple arrays containing individual props if getting the right device tree values into the right places proves tricky.

The meta would look something like this: (pseudocode)

```
meta[0...n] {
    animation:
        Pointer to the animation function

    key:
        the key identifier or NULL if it's not a key LED

    position:
        x: [0;255]
        y: [0;255]

    flags: ?
}
```

Let's take it step by step:

- Animations would be functions. Each could potentially live in its own file and maintain its own state necessary to keep it going depending on what kind of animation it might be - not interfering with any other ones that might be going on. It would be called with the pixel meta and expected to return the pixel values for the new frame.
- Key marks what key the LED belongs to. Which can be used for overriding the default animation. In which case we might need another `alt_animation` property or something?
- Let's assume your keyboard PCB is the reference plane, in which case these would be the relative `x` and `y` coordinates. Keyboards aren't squares though so potentially, somewhere, an `x/y ratio` needs to exist as well to be references by the animations if necessary.
- Any additional flags we might need I guess? Could be useful for dedicated indicator LEDs or to account for any edge cases.

So this is more or less my angle on it, I'd be curious to hear what others think before I start making inroads towards an actual implementation.

joelspadin, 2021-07-15:
I was thinking that rather than having each key hold a pointer to an animation function, you could implement something like an animation driver API and then have each animation be an instance of a driver that implements it. Then you could set up the animations like layers and enable/disable each one individually. Every time you want to update the LEDs, you'd loop through the enabled animations and just call one update function on each, giving it a pixel buffer (and maybe also a list of which pixels it's assigned to update?).

Then you could easily implement things like an "animation" that just takes the output from the previous animation and zeroes out every key that doesn't have a function on the current layer, or one that overrides just the caps key based on the indicator state. As a bonus, using the driver API means each instance of the driver maintains its own state, so you could make much more complex animations like one that listens to key events and lights up the keys you press. I think that would eliminate any need for flags or an `alt_animation` property. Instead of flagging a key as being an indicator, you'd just put a devicetree property on the indicator animation to tell it that key 20 is caps lock.

Animations should also be able to tell the animation system when they need to update. Say you have a solid color animation as the base, an animation that changes based on the layer, and one for indicator LEDs, then you only need to refresh when the layer or indicator state changes. The animation drivers could listen for ZMK events and then call "request animation update" function which queues up a work item to do an animation update.

For animations that smoothly animate instead of just updating on specific events, you wouldn't want to have multiple animations asking the animation system to update on their own timers. If you had two animations requesting updates at different frequencies/phases, you'd update far more often than you need to. To handle this, you could have a global periodic timer, and have the animations' update functions return a Boolean. If any animation returns `true`, then the global timer will schedule another update. If not, the timer will stop and wait for an animation to manually trigger an update. (Would also let us do other optimizations like throttling the update rate when on battery if that ends up being a big power user.)

The-Briel-Deal, 2024-04-02:
Update on this, it looks like there's a implementation here [diff here](https://github.com/moergo-sc/zmk/compare/main...valdur:zmk-glove80:45511285cf551e290225b4e39e17521e5ce7ff7c), this should cover 3 out of the 5 use cases covered above. (atleast for the the glove 80)

I'm going to dive into this tomorrow and see if I can generalize the implementation and make it more general so it can cover other use cases.

My goal is to just make an easy to use interface so its easy to change lighting in a keymap for a first pass.

steynh, 2024-05-23:
I think in general it would be really useful if we could specify an RGB grid for each layer. That way it becomes possible to visually highlight groups of keys (such as a numpad block) in different layers. And then it also becomes very obvious which layer is active. Both of these would be so helpful when just getting used to a new layout.

Animations are cool to have, but they seem quite complex to implement. Whereas just a simple grid per layer seems relatively simple to implement and actually offers a lot of practical benefits.

darknao, 2024-05-23:
> Update on this, it looks like there's a implementation here [diff here](https://github.com/moergo-sc/zmk/compare/main...valdur:zmk-glove80:45511285cf551e290225b4e39e17521e5ce7ff7c), this should [...]

You might also be interested at https://github.com/moergo-sc/zmk/pull/30 which is based on the implementation you mentioned but taken a bit further. It's still specific to glove80, but I think it needs a bit less work to make it available to more boards.

joelspadin, 2024-05-23:
I still think we should provide a generic driver API for setting RGB LED colors with optional animations. We would have the ability to implement "animations" that don't constantly animate, for example with an animation driver that stores a bitmap per layer and requests a new frame any time the layer changes.

My current thoughts on how this might work (a simplified version of my comment from a few years ago):

- The animation driver API would consist of the following functions:
  - "Enable" and "disable" functions to tell a driver that it is/isn't being used.
  - A "render" function which is provided a pixel buffer and returns a value indicating whether the animation should continue.
- Animation control code would have a reference to a "root" animation driver. 
  - Upon enabling lighting, it would call this driver's render function.
  - As long as the render function continues to return `true`, the animation control code will keep rendering on a timer.
  - If the render function returns `false`, the animation control code will stop rendering.
  - Animation drivers can call a "request frame" function to restart rendering.
- Animations could be composed by having one driver's render function call a different driver's render function. This would allow for creating just about any behavior you want from a very simple API. For example (python-ish pseudocode):

```py
# caps lock animation
class CapsLockAnimation(Animation):
    child: Animation
    leds: list[int]
    color: Color

    def render(self, bitmap):
        # Draw the child animation
        continue_anim = child.render(bitmap)
    
        # Then overwrite some pixels if caps lock is on
        if zmk_caps_lock_active():
            for index in self.leds:
                bitmap[index] = self.color
            
        return continue_anim

def on_caps_lock_changed():
    # If the animation is stopped, we need to restart it any time caps lock changes
    zmk_request_animation_frame()

# layer animation
class LayerAnimation(Animation):
    # Set an animation for each layer
    children: list[Animation]

    def render(self, bitmap):
        # Pass through to the animation for the highest active layer
        layers = zmk_keymap_layer_state()
        return self.children[highest_bit(layers)].render(bitmap)

def on_layer_state_changed():
    # If the animation is stopped, we need to restart it any time the layer state changes
    zmk_request_animation_frame()

# solid color animation
class SolidColorAnimation(Animation):
    color: Color 

    def render(self, bitmap):
        # Set every pixel to the same color
        for i in range(len(bitmap)):
            bitmap[i] = self.color
            
        # Colors aren't changing over time, so stop animating
        return False

# color cycle animation
class ColorCycleAnimation(Animation):
    speed: float = 0.1
    hue: int = 0
    last_timestamp: int = 0
    
    def enable(self):
        self.last_timestamp = now()
    
    def render(self, bitmap):
        # Rotate the hue according to the time elapsed since the last frame
        time_elapsed = now() - last_timestamp
        last_timestamp += time_elapsed
        
        hue = (hue + time_elapsed * speed) % 360
        color = hsv_to_rgb(hue, 100, 100)
        
        # Set every pixel to the same color
        for i in range(len(bitmap)):
            bitmap[i] = color

        # Animation runs forever
        return True

# Keyboard-specific setup
layer0 = SolidColorAnimation(color=WHITE)
layer1 = SolidColorAnimation(color=RED)
layer2 = ColorCycleAnimation()

layers = LayerAnimation(children=[layer0, layer1, layer2])

root = CapsLockAnimation(leds=[1], color=YELLOW, child=layers)
```
(For brevity, I've omitted several details, like how an animation's enable function should call its child's enable function, or how the layer switching animation should enable only the child animation for the active layer and disable the others.)

AnthonyNguyen168, 2024-06-04:
I wonder what is making this feature takes so much time. I am a developer and really need this feature for my new keyboard design. Since QMK does not supporting both RGB-per-key and Bluetooth on one PCB.

joelspadin, 2024-06-04:
None of the core maintainers are actively working on this because other features are higher priority. That said, there's nothing stopping anyone from implementing this if they feel strongly enough that it should exist.

The-Briel-Deal, 2024-06-08:
Sorry about not updating this. Looked into this a bit and ended up getting sidetracked. Will hopefully try my hand at this again this weekend.

dbodyas, 2024-06-18:
I'm building a split keyboard and now I'm on stage of wiring leds. What option should I choose or it doesn't matter?
[image]
[image]

joelspadin, 2024-06-18:
If we implement things properly, the order should not matter. For simple things like solid colors, the order doesn't matter at all, or for things like a caps lock indicator, you should be able to choose the specific LED indices that are correct for your keyboard.

For more complex animations, we'd need a way to associate LEDs with physical positions and/or specific keys. The work Pete is doing in #2283 could be useful for this. For example, an animation that does an RGB wave horizontally across the board could simply rotate the hue of each LED according to its X coordinate if we have that information.

gpmontt, 2024-06-26:
I think the project from [Nick Coutsos](https://nickcoutsos.github.io/keymap-editor/) would be a great place to tweak whatever color you want in the appropriate key. 
We need to a new way to define how many led are in the strip in a better way like
```
&led_strip {
    chain-length = <NUMBER_OF_LED>;
};
```
whatever.conf
```
NUMBER_OF_LED=<xx>
```

faleandroid, 2025-09-15:
Is there any news on this feature?

AnthonyNguyen168, 2025-09-15:
> I got this up and running at the time I originally published this, but they were unhappy with the syntax or something in my code, and I lost all interest in pushing the code. It was very easy to integrate.

Did you keep the code or merge request? I can help to contribute if mods are not too difficult.

Retic1337, 2025-09-15:
Yeah, I can try to dig it up in the next day or two. I think the majority of it is in the forked ZMK that I have. I'm not sure how up-to-date that is though. The hard part was incorporating the driver at the time for the IS31FL3733, that was the actual Chip that drove the LEDs on the board I had designed. Otherwise, you basically created a look up table in the DTS config file per board. This would then map a key to a led via a 2d lookup table. From there, you would determine how you want to have the LEDs activated and controlled from part of the software that handles key strokes in my case. So every time a stroke function is called. You could also call out to this backend controlling the LEDs so for instance, set all LEDs to white upon start up and then if a key stroke is detected map the LED from our lookup table to the corresponding key and set that color to red with a countdown timer to turn it off after let's say 500 ms. A big thing to consider when designing a circuit board that will have something like LEDs is power consumption. Sk6812 and equivalent are extremely power hungry. So I would highly recommend against trying to incorporate anything along those lines. It would have to be a LED controlled by some kind of driver, Chip. Like the ones by ISSI. They used very little power.

faleandroid, 2025-09-17:
> In case it's of any help, I published all of my work from [#1046](https://github.com/zmkfirmware/zmk/pull/1046) - and a little extra - [as a Zephyr module](https://github.com/crystalplanet/zmk-rgb-fx). That way it should generally keep working as long as ZMK doesn't break compatibility, with minimum work. I also started moving the drivers from ([#730](https://github.com/zmkfirmware/zmk/pull/730) - completed, [#886](https://github.com/zmkfirmware/zmk/pull/886) - todo) to [crystalplanet/zephyr-led-strip-drivers](https://github.com/crystalplanet/zephyr-led-strip-drivers). But the rgb-fx system is compatible with any Zephyr LED strip driver implementation.
> 
> Unfortunately, I didn't get around to writing docs for it yet, nor did I want to appropriate [@danieldegrasse](https://github.com/danieldegrasse)'s previous work from the PR above. But the system currently works exactly the same, except some names have changed - mostly prefixes.
> 
> Contributions are welcome, happy to review anything that comes in as long as I have something to test it on. One last thing worth noting is I never considered splits when implementing this so there's some work to be done there if you want to have the two halves in sync.
> 
> HTH

I went through your code and also the configuration that you linked at [#1046](https://github.com/zmkfirmware/zmk/pull/1046). But I'm quite a beginner, can u ask you if you can tell me how to use your module with a SK6812MINI-E led strip? Or I must have a led driver to use this implementation?

ice9js, 2025-09-17:
Won’t go into the details here, but you can find my latest config that’s using the driver and rgb-fx modules here: https://github.com/crystalplanet/zmk-config/tree/onigiri/config/boards/arm/onigiri

Couple of notes:
- In the rgb-fx node you’ll see ‘key-pixels’ is empty, that’s because that keyboard doesn’t actually use any key led’s. By default, it assumes the pixels in the LED strip are in the same order as your physical key definitions, but you can use that setting to override it.
- yes, it would also work with sk6812, I think Zephyr even supports that out of the box. Just put your led strip instance into ‘drivers’ and update ‘chain-lengths’. It’s also possible to use several led strip drivers in combination - the system provides another level of absteaction on top of it so you’re not limited by the electrical board layout.

For any more support/details, I’d suggest to either open up issues in the rgb-fx repo or ping me on the zmk discord. To prevent confusion in the future as this is not really officialy supported by ZMK.

gudzpoz, 2026-10-08:
Recently, I've also implemented a [`zmk-rgb-matrix`] module[^slop], which has gone through quite some refactoring. Its current design mostly follows the design proposed by @joelspadin: demand-driven rendering, each effect as a driver, (partially synchronized) split support, and a bit more. During the trial-and-error, there are a few interesting designs, feedback, and caveats that I find worth discussing, so here's my two cents:

- On specifying the positions of the LEDs: For per-key LEDs, the `zmk,physical-layout` (implemented in the mentioned PR #2283) is *very useful*, as it (mostly) frees the user from having to specify the position of every single LED. But users will still need to describe:
  - LED-chain order (as is exemplified by @dbodyas's comment 🥲),
  - and, for non per-key LEDs (like standalone status LEDs, which I do hope ZMK to support), their positions in `zmk,physical-layout` coordinate system.

  Currently, `zmk-rgb-matrix` uses the following devicetree "syntax", which gets the job done but probably needs more polish and discussion:

  ```dts
  rgb_matrix: rgb-matrix {
    compatible = "keypaw,rgb-matrix";
    strip = <&led_strip>;
    physical-layout = <&default_layout>;
    mapping = <
      5 4 3 2 1 0             // Keys, with positions specified in &default_layout
      KP_RGB_NO_KEY(50, 387)  // Standalone LED, with manually specified position
    >;
  }
  ```

- On multiple strip support: A previous PR (#1046) actually have support for multiple LED strips, which my original design also intended to follow. But, it seems to me now it's probably cleaner to have a dedicated LED strip "composite" driver:

  ```dts
  strip1: ws2812-1@0 { compatible = "worldsemi,ws2812-spi"; chain-length = <3>; /* ... */ };
  strip2: ws2812-2@0 { compatible = "worldsemi,ws2812-spi"; chain-length = <2>; /* ... */ };
  led_strip: led_strip {
    compatible = "zmk,led-strip-composite";
    strips = <&strip1 &strip2>;
    chain-length = <5>;
  };
  ```

  This should keep board wiring more separate from the rendering API.

  > However, though, we will still need to support zero strip builds to support dongles.

- Split support: There are different "levels" of split support.

  1.  Do halves always show the same kind of animation?
  2.  Is the phase of animations (breathing, rainbow, etc.) guaranteed to be in sync?
  3.  Should halves be under the same coordinate system? (Like, should there be two rainbows, one per half, or a single rainbow, with each half showing half of it?)
  4.  Do reactive animations travel across halves? (Like, should the ripples from ripple animations travel across halves?)

  The design of `zmk-rgb-matrix` synchronizes the selected effect, and periodically synchronizes a "phase" clock for time-based effects. It can't guarantee 1 and 2 though, as you can't prevent a user from shutting off peripheral halves and then adjusting the settings, so one might need to check for peripheral connections and re-synchronize whenever a new peripheral connects.

  For 3 and 4, I chose to leave geometry and reactive feedback half-local. (IMO, because splits are separated physically, with the distance/angle between them entirely decided by where the user puts them, it doesn't make much sense to have them share the same coordinate system. However, a shared logical layout could still be useful for continuous animations.)

  But opinions can vary, and I think these should definitely be explicitly discussed and decided upon, since they can have quite an impact the API and architectural design.

- Ensuring flexibility and ease-of-use: This is my favorite part of the current `zmk-rgb-matrix` design. Basically, it follows the "stacked transforms" model proposed by Joel, separating things into three components:

  - Effects: generate pixel colors.
  - Conditions: reusable boolean predicate, reevaluated on source changes or scheduled deadlines.
  - Overlays: conditionally transform the existing output over selected LEDs.

  Each of them is a Zephyr device, so it is extensible from 3rd party modules, and also easy to customize by users using only devicetree properties. Separating condition evaluation from rendering also means indicators and control conditions can keep updating and synchronized while a static effect needs no further frames or while lighting is off.

  Also, we support nesting effects inside an overlay devicetree definition. Building on that, we can have a built-in overlay implementation (`keypaw,rgb-overlay`) that accepts a child effect, and overlays the effect onto the pixels (`all-leds` or a user-specified list of LEDs):

  ```dts
  ble_p0_off: ble_p0_off {
    compatible = "keypaw,rgb-overlay";
    condition = <&cond_ble_0_connecting>;
    keys = <5 6>;
    fx_p0_off {
      compatible = "keypaw,rgb-matrix-breathe";
      #binding-cells = <0>;
      mode = "brightness";
      color = <0x00FF00>;
    };
  };
  ```

  And building on this and conditions, we can have:

  | Condition | Child effect | Result |
  |-----------------------------|--------------|-------------------------------------|
  | CapsLock on | Solid | CapsLock indicator |
  | BLE (profile #0) connecting | Breathing | Breathing LED under your BLE hotkey |
  | BLE (profile #0) connecting | Rainbow | Rotating-hue LED under the BLE key |
  | Layer active | Any | Per-layer animation |

[... cut ...]
I'd especially appreciate thoughts on synchronization between splits, as the way that `zmk-rgb-matrix` implements it (and other ways I can come up with) is not very elegant. For example, it polls the source id list periodically to detect whether new peripherals have connected, and I'm curious if there's a simpler way.

[^slop]: Please note that (1) it's mostly AI generated code, though I tried my best to keep the entropy low, and (2) it is not developed under ZMK's clean-room requirements (I consulted QMK code, and, used AI). So the code should not be used as a reference for an upstream ZMK implementation.

[`zmk-rgb-matrix`]: https://github.com/gudzpoz/zmk-rgb-matrix