zmkfirmware/zmk #560: Github authentication method deprecation

keyboard-magpie, 2020-12-31:
[image]

When using the basic setup scripts (https://zmkfirmware.dev/docs/user-setup - curl and wget methods)
it appears we will soon run into the above issue meaning the only way to authorise will be via tokens- 

I propose the docs need amending to reflect this

joelspadin, 2020-12-31:
We should probably have the docs point to https://docs.github.com/en/free-pro-team@latest/github/authenticating-to-github/creating-a-personal-access-token (and optionally https://docs.github.com/en/free-pro-team@latest/github/authenticating-to-github/connecting-to-github-with-ssh since SSH keys should still work too).

caksoylar, 2023-11-05:
Docs no longer go into detail of setting up git access and point to Github docs, so this no longer seems applicable.

closed 2023-11-05 by caksoylar