zmkfirmware/zmk #562: Develop header/copyright checker

innovaker, 2021-01-01:
Files with missing or incorrect headers are creeping through the review process.  This gradually builds up a maintenance burden.

Let's implement a guard/check to lighten the load.

**MUST**:
- be able to run in GitHub workflows (as a guard)

**MAY**:
- reuse existing tools
- be able to run locally
  - via `pre-commit` or a `git` hook perhaps?
  - _stretch goal!_

innovaker, 2021-01-02:
This task is open to anyone.  It's also a good task for someone who wants to contribute to ZMK but doesn't have C skills.

innovaker, 2021-01-06:
An architecture/approach similar to [commitlint](https://github.com/conventional-changelog/commitlint) with [commitlint-github-action](https://github.com/wagoid/commitlint-github-action) may be a good starting point.

e11i0t23, 2021-01-31:
Hi, I am going to take a look and do some work on this

e11i0t23, 2021-02-02:
I Am going to work on the github workflow next however here is a simple git hook I put together that works locally to achieve the same thing https://gist.github.com/e11i0t23/71ccdf7cd78027832669665bcce7de39

e11i0t23, 2021-02-03:
Hi, ive set this us https://github.com/e11i0t23/zmk/blob/headercheck/headerCheck.py with github actions to run this, it checks all newly added files in comparison to main, example outputs: success - https://github.com/e11i0t23/zmk/runs/1819059230?check_suite_focus=true
failure - https://github.com/e11i0t23/zmk/runs/1818995029?check_suite_focus=true
currently i have it set up to ignore yaml, md and json files, let us know anything you would like adjusting ect before i submit any sort of pull request

innovaker, 2021-02-04:
Many thanks @e11i0t23, this is a good start!

Please may I check whether you looked into any pre-existing tools before working on this?

---

> Also are we wanting this to only check files added in app/ or the whole repo?

Added is fine.  We should probably also check files which are modified.

That said, having it flexible enough to run on an arbitrary folder/repo (from a CLI?) could be a handy bonus for developers if it were to be published as a python package.  I appreciate that's another layer up in complexity.  commitlint or other related tools may be a source of inspiration for that *if* you wish to do so.

---

Although checking the copyright name (primarily `The ZMK Contributors`) and license identifier is the crux of the issue, it may also be useful to validate the whole header format as well.  Sometimes people fail to copy and paste it correctly, or have the wrong date, or throw in (or delete) random spaces or characters.  If you can imagine it, someone will probably do it at some point.

May I suggest considering a regex approach so that the whole format of the header can be checked?  I did some preliminary work around this when I last tidied stuff up.  Please see these commits (messages) if you feel that's worth exploring:
- f17584ca0ee4d6bbc0893c0a62a0dc2fa293d2b8
- 02614b6c859fa284d48e223765c58f18a0cef281

---

Is this something you're looking to publish separately for general use or only use in ZMK?  Either is fine.

e11i0t23, 2021-02-05:
> Please may I check whether you looked into any pre-existing tools before working on this?

Yes, I looked at things like what you linked and https://github.com/marketplace/actions/license-checker, however that was not featured correctly for what was I understand we are wanting. And then as I started just trying to write it as a simple pre-commit git hook that's why I made it as a standalone thing. It also enable ease of running locally as is simply a python script, something everyone installs when following the getting started dev guide

Regarding running locally, I see we added pre commit hooks package in pr #386 so i will have a look at how to implement this to run with that,

---

> May I suggest considering a regex approach so that the whole format of the header can be checked? I did some preliminary work around this when I last tidied stuff up. 
> Added is fine. We should probably also check files which are modified.

These sounds sensible changes to add I will have at look at doing this

---

> Is this something you're looking to publish separately for general use or only use in ZMK? Either is fine.

This was just made for use in ZMK and would be intended to be merged into main branch (the .py file and .yaml for the workflow) once its to the features as needed. I can change the copyright header to `The ZMK Contributors` if that is the preferred header

innovaker, 2021-02-10:
> however that was not featured correctly for what was I understand we are wanting

Sorry, I wasn't suggesting that the tools themselves were a solution for the problem, I meant that their archtechture/approach was worth considering.  i.e. a script designed to run as CLI but can integrate into hooks/pre-commit/husky, GH actions etc.

---

> These sounds sensible changes to add I will have at look at doing this

Thanks!

---

> This was just made for use in ZMK and would be intended to be merged into main branch

Sounds great!  If you wanted to do so, this type of tool could be generalised quite easily for any project to use.

---

> I can change the copyright header to The ZMK Contributors if that is the preferred header

The two header formats we use (depending on file format) are:
```
/*
 * Copyright (c) 2021 The ZMK Contributors
 *
 * SPDX-License-Identifier: MIT
 */
```

```
# Copyright (c) 2021 The ZMK Contributors
# SPDX-License-Identifier: MIT
```

The date currently varies based on when it was created.

e11i0t23, 2021-02-14:
@innovaker 

Updated to regex style approach to check whole header see line 46 and 47 of https://github.com/e11i0t23/zmk/blob/headercheck/headerCheck.py for the regex I am using

---

I Have now also added checks in for modified files, for these a data of 2020 is still accepted, whereas for a new file the current year must still be in them

---

please see line#7 for which files ive seen to include the first header style from your previous message and which file types im ignoring 

--- 

borked git history on this branch so will make a new one before a PR, any other changes needed as well before this is suitable?

Quard, 2023-02-22:
Hello,

I have started working on this issue by checking existing solutions. But what I have found does not meet the requirements, at least from my understanding.

I have created pre-commit plugin that also could be used separately as CLI script as well. This script could support different types of comments and validate content of this comment using regular expression. 

Below you can find example of execution on current codebase of ZMK with excluded app/boards folder and example of configuration that was used.

[image]

```yaml
  - repo: ../license-header-checker
    rev: 30ffa2d80698a4b57425ead20c15e6f5d3265cc5
    hooks:
      - id: license-header-checker
        files: \.(c|h|dtsi|dts|overlay|keymap)$
        exclude: ^app\/boards\/
        args:
          - --comment-style
          - "/*| *| */"
          - --license
          - "\\nCopyright \\(c\\) \\d{4}(-\\d{4})? The ZMK Contributors\\n\\nSPDX-License-Identifier: MIT\\n"
      - id: license-header-checker
        files: \.(yaml)$
        exclude: ^app\/(boards|tests)\/
        args:
          - --comment-style
          - "#"
          - --license
          - "Copyright \\(c\\) \\d{4}(-\\d{4})? The ZMK Contributors\\nSPDX-License-Identifier: MIT"
```

If this solution looks good to you, I will create pull-request for integration after finishing my script, still need to increase test coverage and write some documentation.