zmkfirmware/zmk #575: document commit guidelines

innovaker, 2021-01-03:
Let's develop a ZMK orientated simplified summary of [conventional commits](https://www.conventionalcommits.org/en/v1.0.0/).

It should *concisely* explain:
- why use conventions?
- the commit message format
- an example commit
- title/subject
  - a list of types, and when to use each
  - a list of scopes, and when to use each
- body/description
  - what should be included
- footers
  - formats
  - list of possible footers
 - FAQs?

Whoever takes on this issue, please contact me before starting development.

Until then, let's use this issue to collate together possible types, scopes and footers.

innovaker, 2021-01-03:
#### Footers
- `Closes: #xxx` - closes GitHub issue.
- `See: #xxx` - see related GitHub issue or PR.
- `See: b82391a408edd8b933eab595993dd4d39320a647` - see related issue.

mcrosson, 2021-01-03:
Should it be documented that multiple commits are good for larger change's that are all interwoven/inter dependent? ie. For something like a shield overhaul there should be multiple commits with the different changes made.

Any thoughts about squashes ahead of a merge being needed/desired.

innovaker, 2021-01-03:
I guess we could include a very concise guide on commit best practices, but keep it short and link to one of the many guides that are available.

mcrosson, 2021-01-08:
looks like the following are most heavily used in git already for conventional commits

List pulled from `git log --format=oneline | grep -o "[a-z]\+([^)]\+)" | sort | uniq -c`

Types:
  - fix
  - feat
  - docs
  - refactor

Scopes:
  - Docs
  - [Shield]
  - [Feature]
  - [Zone]

mcrosson, 2021-01-08:
Below is a code snippet that will setup commitlint+husky as a local git hook for `commit-msg` and verification commit messages follow standard conventional commit semantics. This is meant to be run _locally_ and tested with the current dev container.

``` sh
# cd /path/to/zmk
npm install --save-dev @commitlint/{cli,config-conventional}
echo "module.exports = {extends: ['@commitlint/config-conventional']};" > commitlint.config.js
npm install --save-dev husky@next
npm install pinst --save-dev
npx husky install
npx husky add .husky/commit-msg "npx --no-install commitlint --edit $1"
npx commitlint --from HEAD~1 --to HEAD --verbose
touch foo
git commit -m "foo: this will fail"
git commit -m "chore: lint on commitmsg"
git reset HEAD~1
rm foo
```

nmunnich, 2025-08-04:
We both document and enforce conventional commits now.

closed 2025-08-04 by nmunnich