zmkfirmware/zmk #614: How to add more gpios?

xiaoquanaa, 2021-01-18:
Hi, guys, i watched nicell's video of zmk on youtube , it helped me a lot in making my self 's first bluetooth customed key board . but what confused me is that the zmk's gpio number seems fixed in 21(which is pro mico's gpio number) . yes it is enough to assemble a 100key keyboard by a 10x10 matrix with matrix-transform , but this lead to a uncomfortable feeling when routering the pcb . so i think maybe there is some way to add the gpio number to 25 or more , thus i can make a 19x6 matrix . but since i am not a professional programmer , and English is not my first language , so i have trouble in doing that .i have found this two files \zmk\app\boards\arm\nice_nano\arduino_pro_micro_pins.dtsi and \zmk\app\dts\common\arduino_uno_pro_micro_map.dtsi maybe the  relative file . i will  appreciate  if  you can spare your time to tell me how to add the gpio number . Thanks !

crides, 2021-01-18:
This is not dependent on the firmware, it's a limitation of the hardware itself. The Pro micro only has 18 GPIOs, and the nice!nano has 21 in total (23 if you also use the 2 SMD pads). So unless you are using another board that Zephyr supports and has more IOs, or integrate a module (like the [holyiot](https://www.aliexpress.com/item/32868002366.html)) or a bare chip (that Zephyr supports) into your keyboard PCB it's not possible.

Synless, 2021-07-21:
> This is not dependent on the firmware, it's a limitation of the hardware itself. The Pro micro only has 18 GPIOs, and the nice!nano has 21 in total (23 if you also use the 2 SMD pads). So unless you [...]

Wouldn't it be possible to use a shield with shift registers to achieve more IO ?

crides, 2021-07-21:
Of course, but using a dedicated IO expander would probably be easier

Synless, 2021-07-21:
> Of course, but using a dedicated IO expander would probably be easier

Thanks for the swift answer.
What would be the best way to work with shift register ? Does ZMK supports is natively or a "manually written key scan" would be needed ?

crides, 2021-07-21:
I'm not sure why one would actually use shift registers for this purpose. There's support for IO expanders in ZMK

Synless, 2021-07-21:
Could you link to an example of that ?
It the documentation of ZMK only what is on [https://zmk.dev/docs](https://zmk.dev/docs) ?
I struggle to find GPIO read/write and the regular MCU interactions.

crides, 2021-07-22:
You should check the [Ferris PR](https://github.com/zmkfirmware/zmk/pull/642)

Until ZMK has support for pluggable modules, you'll have to modify the internals. As it's based on Zephyr, all the HAL utilities comes from there.

caksoylar, 2022-04-03:
Closing as it is a user question, as an aside `#hardware` channel at the [ZMK Discord](https://zmk.dev/community/discord/invite) is a useful place for such questions and discussions.

closed 2022-04-03 by caksoylar