zmkfirmware/zmk #631: Early boot recognition in linux

Eyenseo, 2021-01-24:
I'm using zmk+nice!nano in a split-kb and have an issue with booting my linux system - in the early boot stages there is no bluetooth, but I need one to input my decryption password. So I added the necessary modules (https://github.com/irreleph4nt/mkinitcpio-bluetooth) but the keyboard is not responsive/available/recognised(?). Does anyone have experiences with a similar setup, or ideas what I'm missing?

[ergodash_linux_bootloader.log](https://github.com/zmkfirmware/zmk/files/5861886/ergodash_linux_bootloader.log) from boot of computer, keyboard tries to connect but doesn't succeed. I made sure that BLE is preferred and that my initramfs has the right pairing keys.

tominabox1, 2021-01-24:
I am not sure but I'm willing to guess you'd be suffering from the same problem as dual booting on 2 different OSs. ZMK only pairs with a physical BT MAC once, therefore the keys will not match when you're in your bootloader. You may be able to manually alter the MAC of your BT device in the bootloader on Linux (or the one in the OS) so that it "looks" like your computer has 2 BT devices then you can pair your keyboard to both of them on separate profiles.

Eyenseo, 2021-01-24:
I would argue against that, because you copy all information over. And other BT devices would face the same problem. But if you can tell me how to figure that out I'll have a look :)

Eyenseo, 2021-11-30:
Just saw that this was still open - it was a missing module iirc. I think it was `uhid`

closed 2021-11-30 by Eyenseo