zmkfirmware/zmk #632: Boot protocol not enabled

xudongzheng, 2021-01-24:
Boot protocol may be necessary to use as USB keyboard with certain BIOS. Consider adding `CONFIG_USB_HID_BOOT_PROTOCOL=y`.

xudongzheng, 2021-02-04:
Typically when sending HID data, the first byte is the report ID. That may need to be excluded when sending a boot report. It's also possible that boot reports are sent in a completely different manner.

typorian, 2022-07-28:
Just to clear up what this question is about, I hadded the config switch you mentioned to my config but I still can't use my kyria with nice!nano v2 to enter the bitlocker pin. Is this expected behavior waiting for a fix or am I doing it wrong? Thank you!

xudongzheng, 2022-07-28:
> Just to clear up what this question is about, I hadded the config switch you mentioned to my config but I still can't use my kyria with nice!nano v2 to enter the bitlocker pin. Is this expected behavior waiting for a fix or am I doing it wrong? Thank you!

No, that configuration is just step one of many steps needed to implement boot protocol. There is https://github.com/zmkfirmware/zmk/pull/1140 but that's unfortunately a bit outdated.

xudongzheng, 2022-09-15:
Closing in favor of #913.

closed 2022-09-15 by xudongzheng