zmkfirmware/zmk #665: Refactor battery_voltage_divider Kconfig to not select ADC

Nicell, 2021-02-04:
Currently the `battery_voltage_divider` sensor selects `ADC`, which prevents us from putting our sensor Kconfig in an if block.

It should be refactored so that it only depends on it, and then boards should enable `ADC` as they currently do for `GPIO`. We'll only need nRF52 boards to enable `ADC` though as currently the driver doesn't support other types of ADC.

petejohanson, 2022-09-29:
@Nicell Is this really needed any more now that we default this based on the DTS, e.g. `default $(dt_compat_enabled,$(DT_COMPAT_ZMK_BATTERY_VOLTAGE_DIVIDER))` ?

myst729, 2023-11-21:
Is this refactor eventually aiming for a generic ADC driver? I'm looking forward to seeing that happen, as I would like to try plenty of analog peripherals, such as joystick, linear slider, hall effect sensor, etc.