zmkfirmware/zmk #686: Keys in combos and mod-tap getting stuck

jreiter, 2021-02-18:
Board: nice!nano
Shield: QAZ
User repo: https://github.com/jreiter/zmk-config/tree/qaz

The firmware I was using at the time that I most recently reproduced this was built using Github actions here: https://github.com/jreiter/zmk-config/actions/runs/578609025

Sometimes when I'm typing on my QAZ with nice!nano using ZMK, I've noticed some keys get stuck.  So far, I've only seen this happen with keys that are involved in a combo or mod-tap - primarily shift (mod-tap in my keymap) and tab (combo in my keymap).  It's pretty easy for me to reproduce by doing typing tests, so it seems to primarily happen while typing as fast as possible (about 90-100 wpm for me).  Once the key is stuck, the only way I've been able to figure out to get my board working again is to get to bootloader and reflash.

tio logs attached from the last time I made this happen, when tab got stuck.
[tio.log](https://github.com/zmkfirmware/zmk/files/6005385/tio.log)

mcrosson, 2021-02-18:
I'm seeing similar if using combos/hold-tap sequences in quick succession. it's not consistent as i don't type too fast with the below setup but it happens just enough where it's likely the same bug for me as the original reporters

keymap / implementation in use:

https://git.kemonine.info/kemonine/keyboard/src/branch/master/zmk-config/config/artsey_v0_7.dtsi

https://git.kemonine.info/kemonine/keyboard/src/branch/master/zmk-config/config/boards/shields/corne_artsey_left

jreiter, 2021-02-18:
Some more info that I realized was relevant after discord discussion - when tab (combo) got stuck, I was unable to make any other key presses register.  When shift (mod-tap) got stuck, I was still able to use my reset mapping.

okke-formsma, 2021-02-18:
Thanks for the report!

Unfortunately I can't locate the TAB keypress (code 0x2B, position 37 or 39) in the logs.

Could you increase the `LOG_BUFFER_SIZE` kconfig parameter to 32000 or something like that to prevent missing logs? Also, I've made a ZMK branch that adds additional logging to combos. Could you try to reproduce with that? (https://github.com/okke-formsma/zmk/tree/combos-logging). To do this, change west.yml in your config directory as follows:

```
manifest:
  remotes:
    - name: zmkfirmware
      url-base: https://github.com/zmkfirmware
    - name: okke
      url-base: https://github.com/okke-formsma
  projects:
    - name: zmk
      remote: okke
      revision: combos-logging
      import: app/west.yml
  self:
    path: config
```

azor738, 2021-02-20:
I can reproduce the issue of keys getting stuck by holding one key in the combo and tapping another, over bluetooth or USB, for Windows 10 on 3 PCs and also in Android over both bluetooth and USB.

Here's the relevant log section for holding J and tapping semicolon, results in a stream of "ssssssssss" (because it's a Dvorak layout the semicolon gets translated to s)

[00:05:33.716,735] <dbg> zmk.kscan_gpio_read_0: Sending event at 2,1 state on
[00:05:33.716,796] <dbg> zmk.zmk_kscan_process_msgq: Row: 2, col: 1, position: 31, pressed: true

[00:05:33.716,827] <dbg> zmk.position_state_changed_listener: 31 bubble (no undecided hold_tap active)
[00:05:33.716,857] <dbg> zmk.zmk_event_manager_handle_from: Listener captured the event
[00:05:33.901,763] <dbg> zmk.kscan_gpio_read_0: Sending event at 2,4 state on
[00:05:33.901,824] <dbg> zmk.zmk_kscan_process_msgq: Row: 2, col: 4, position: 34, pressed: true

[00:05:33.901,855] <dbg> zmk.position_state_changed_listener: 34 bubble (no undecided hold_tap active)
[00:05:33.901,885] <dbg> zmk.zmk_event_manager_handle_from: Listener captured the event
[00:05:33.916,931] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 31, binding name: KEY_PRESS
[00:05:33.916,961] <dbg> zmk.on_keymap_binding_pressed: position 31 keycode 0x7000d
[00:05:33.917,022] <dbg> zmk.hid_listener_keycode_pressed: usage_page 0x07 keycode 0x0d implicit_mods 0x00 explicit_mods 0x00
[00:05:33.917,022] <dbg> zmk.zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[00:05:33.917,022] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:05:33.917,114] <dbg> zmk.position_state_changed_listener: 34 bubble (no undecided hold_tap active)
[00:05:33.917,144] <dbg> zmk.zmk_event_manager_handle_from: Listener captured the event
[00:05:33.979,309] <dbg> zmk.kscan_gpio_read_0: Sending event at 2,4 state off
[00:05:33.979,370] <dbg> zmk.zmk_kscan_process_msgq: Row: 2, col: 4, position: 34, pressed: false

[00:05:33.979,400] <dbg> zmk.position_state_changed_listener: 34 bubble (no undecided hold_tap active)
[00:05:33.979,400] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 34, binding name: KEY_PRESS
[00:05:33.979,431] <dbg> zmk.on_keymap_binding_pressed: position 34 keycode 0x70033
[00:05:33.979,492] <dbg> zmk.hid_listener_keycode_pressed: usage_page 0x07 keycode 0x33 implicit_mods 0x00 explicit_mods 0x00
[00:05:33.979,492] <dbg> zmk.zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[00:05:33.979,492] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:05:33.979,583] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 34, binding name: KEY_PRESS
[00:05:33.979,614] <dbg> zmk.on_keymap_binding_released: position 34 keycode 0x70033
[00:05:33.979,644] <dbg> zmk.hid_listener_keycode_released: usage_page 0x07 keycode 0x33 implicit_mods 0x00 explicit_mods 0x00
[00:05:33.979,644] <dbg> zmk.zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[00:05:33.979,675] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:05:34.057,952] <dbg> zmk.kscan_gpio_read_0: Sending event at 2,1 state off
[00:05:34.058,013] <dbg> zmk.zmk_kscan_process_msgq: Row: 2, col: 1, position: 31, pressed: false

[00:05:34.058,044] <dbg> zmk.position_state_changed_listener: 31 bubble (no undecided hold_tap active)
[00:05:34.058,074] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 31, binding name: KEY_PRESS
[00:05:34.058,105] <dbg> zmk.on_keymap_binding_released: position 31 keycode 0x7000d
[00:05:34.058,135] <dbg> zmk.hid_listener_keycode_released: usage_page 0x07 keycode 0x0d implicit_mods 0x00 explicit_mods 0x00
[00:05:34.058,135] <dbg> zmk.zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[00:05:34.058,166] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07

************ I press the stuck key again, the keypress is missed on Android (which filters the repeated keypresses), windows stops repeating

[00:05:53.844,970] <dbg> zmk.kscan_gpio_read_0: Sending event at 2,4 state on
[00:05:53.845,031] <dbg> zmk.zmk_kscan_process_msgq: Row: 2, col: 4, position: 34, pressed: true

[00:05:53.845,062] <dbg> zmk.position_state_changed_listener: 34 bubble (no undecided hold_tap active)
[00:05:53.845,092] <dbg> zmk.zmk_event_manager_handle_from: Listener captured the event
[00:05:53.911,712] <dbg> zmk.kscan_gpio_read_0: Sending event at 2,4 state off
[00:05:53.911,804] <dbg> zmk.zmk_kscan_process_msgq: Row: 2, col: 4, position: 34, pressed: false

[00:05:53.911,804] <dbg> zmk.position_state_changed_listener: 34 bubble (no undecided hold_tap active)
[00:05:53.911,834] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 34, binding name: KEY_PRESS
[00:05:53.911,865] <dbg> zmk.on_keymap_binding_pressed: position 34 keycode 0x70033
[00:05:53.911,895] <dbg> zmk.hid_listener_keycode_pressed: usage_page 0x07 keycode 0x33 implicit_mods 0x00 explicit_mods 0x00
[00:05:53.911,926] <dbg> zmk.zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[00:05:53.911,926] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:05:53.912,017] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 34, binding name: KEY_PRESS
[00:05:53.912,048] <dbg> zmk.on_keymap_binding_released: position 34 keycode 0x70033
[00:05:53.912,078] <dbg> zmk.hid_listener_keycode_released: usage_page 0x07 keycode 0x33 implicit_mods 0x00 explicit_mods 0x00
[00:05:53.912,078] <dbg> zmk.zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[00:05:53.912,109] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07

This is the combo definition:

combo_numeric {
    timeout-ms = <200>;
    key-positions = <25 28 31 34>;
    bindings = <&tog NUM>;
};

Sorry Okke, couldn't get the west.yml file to work on Github, rejects the workflow due to "No event triggers defined in `on`", if I enter the on: [push, pull_request, workflow_dispatch], new error is "No jobs defined in `jobs`", then I'm lost ┐(´～｀)┌

okke-formsma, 2021-02-24:
@azor738 thanks for the logs. Do you have any hold-taps defined? From the logs I suppose the keys you press are not hold-taps?

azor738, 2021-02-24:
You're correct @okke-formsma, all hold-taps were disabled when I recorded this log

okke-formsma, 2021-02-27:
```

[00:00:23.139,007] <dbg> zmk.kscan_gpio_read_0: Sending event at 1,2 state on
[00:00:23.139,007] <dbg> zmk.kscan_gpio_read_0: Sending event at 1,4 state on
[00:00:23.139,068] <dbg> zmk.zmk_kscan_process_msgq: Row: 1, col: 2, position: 11, pressed: true
[00:00:23.139,099] <dbg> zmk.position_state_changed_listener: bubble pos 11 (no undecided hold_tap active)
[00:00:23.139,129] <dbg> zmk.zmk_event_manager_handle_from: Listener captured the event
[00:00:23.139,129] <dbg> zmk.zmk_kscan_process_msgq: Row: 1, col: 4, position: 13, pressed: true
[00:00:23.139,160] <dbg> zmk.position_state_changed_listener: bubble pos 13 (no undecided hold_tap active)
[00:00:23.139,190] <dbg> zmk.zmk_event_manager_handle_from: Listener captured the event
[00:00:23.210,815] <dbg> zmk.kscan_gpio_read_0: Sending event at 1,4 state off
[00:00:23.210,876] <dbg> zmk.zmk_kscan_process_msgq: Row: 1, col: 4, position: 13, pressed: false
[00:00:23.210,906] <dbg> zmk.position_state_changed_listener: bubble pos 13 (no undecided hold_tap active)
[00:00:23.210,906] <dbg> zmk.release_pressed_keys: Combo: release first captured event and reraise the rest.
[00:00:23.210,937] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 11, binding name: MOD_TAP
[00:00:23.210,968] <dbg> zmk.on_hold_tap_binding_pressed: 11 new undecided hold_tap
[00:00:23.210,968] <dbg> zmk.on_hold_tap_binding_pressed: 79 tapping term left
[00:00:23.210,998] <dbg> zmk.position_state_changed_listener: ev timestamp 4294990585 sum 257698037761
[00:00:23.211,029] <dbg> zmk.position_state_changed_listener: 11 capturing pos 13 down event
[00:00:23.211,029] <dbg> zmk.zmk_event_manager_handle_from: Listener captured the event
[00:00:23.211,029] <dbg> zmk.zmk_keymap_apply_position_state: layer: 0 position: 13, binding name: MOD_TAP
[00:00:23.211,059] <err> zmk: ACTIVE_HOLD_TAP_CLEANED_UP_TOO_EARLY
```

In some cases combo keys are re-raised so hold-taps can retrigger properly.
The hold-tap can then capture that key until it can make a decision.
We weren't reraising the key-ups, so those would sometimes be executed before the key-down was released by the hold-tap, causing a hanging key.

PR coming soon.

PR #697 (2021-03-11): fix(combos): Fix stuck keys when pressing long combos.
To properly retrigger hold-taps when a combo is _not_ activated, some position down events need to be reraised instead of released. The corresponding position up events were never reraised, causing a potential stuck key. The hold-tap would catch the reraised position down event, while the key up is processed normally. Once the hold-tap releases the position down event, the position up event has already occurred and we've got a stuck key.

tl;dr this fix makes sure position down and position up events are not reordered, preventing a stuck k [... cut]
closed 2021-03-11 by petejohanson