zmkfirmware/zmk #689: Macros v2

okke-formsma, 2021-02-20:
Let's continue the discussion from https://github.com/zmkfirmware/zmk/pull/685 here.

okke-formsma, 2021-02-20:
> @joelspadin said:
> If I correctly understand that you mean &press, &release, and &wait would be special things interpreted only by macros and not regular behaviors that you could also stick into a keymap, I think that would work well. You could also C macro your ZMK macros to get something like &kp:
> 
> ```#define MACRO_KP(key, duration) <&press (key)>, <&wait (duration)>, <&release (key)>```
> 
> That does mean that you wouldn't be able to do things like put an RGB underglow command in the middle of a macro though, but I'm not sure how useful that would be.
> 
> The thing I mainly didn't like about the previous proposal was that it was taking regular keymap behaviors but then needing to layer special macro things on top of every behavior. e.g. &kp should be pressed, wait for X milliseconds, then released, &press and &release should just be pressed (and maybe immediately released?) and continue with no delay, &wait isn't pressed at all, and I have no idea what should happen if you use a &mt...
> 
> I vote for either:
> 
>  *   Macros use their own set of commands that are independent from keymap behaviors.
>  *   Macros use keymap behaviors, but we always execute each behavior the same way (press, wait, release) and have some means to control how long each one is held.
> 
> 

One of the strenghts of ZMK is the flexible keymap system using behaviors. Whatever we do with combos, we should support any and all behaviors from combos. This means option 1 (macros use their own set of commands) is a no-go in my opinion.

In option 2 you state that 'we always execute each behavior the same way (press, wait, release)'. I think we can make that work with an additional rule: wait is only executed when a behavior defines both the 'press' and 'release' bindings. Consider `&press` which is a behavior that only has a "press" binding, and `&release`, a behavior that only defines a "release" binding.  This allows the user to fully in control timing when using `&press` and `&release`, and also any new (user-defined) behaviors in the future.

The `&wait` must be a little special, when the macro encounters a `&wait` it should not call this behavior but only execute a "wait" with the specified delay. I don't see any way to solve this wait within a behavior without blocking the main thread.

to summarize, I think we should:
* leverage the power of the devicetree in macros, which lets us use all existing behaviors
* add behaviors like `&press` and `&release` which could work inside and outside of macros
* add a behavior like `&wait` which is a no-op in a normal keymap but introduces a wait inside a macro

BrainWart, 2021-02-20:
> One of the strengths of ZMK is the flexible keymap system using behaviors. Whatever we do with macros, we should support any and all behaviors from macros. This means option 1 (macros use their own set of commands) is a no-go in my opinion.

I agree that behaviors are flexible and awesome. I don't think the device tree will 

> In option 2 you state that 'we always execute each behavior the same way (press, wait, release)'. I think we can make that work with an additional rule: wait is only executed when a behavior defines [...]

I'm okay with this personally.

> The `&wait` must be a little special, when the macro encounters a `&wait` it should not call this behavior but only execute a "wait" with the specified delay. I don't see any way to solve this wait within a behavior without blocking the main thread.

I don't like the idea of the 'wait' being special. What if all behaviors had a way to report when they finished? The 'wait' behavior could just be a slow 'none'. This would have the side effect of allowing embedded macros when they become asynchronous.

joelspadin, 2021-02-21:
> What if all behaviors had a way to report when they finished?

I don't know exactly how that would be implemented or what implications that might have for regular keymap handling, but that seems like a good solution to me.

okke-formsma, 2021-12-16:
Some chats today on discord about the status of macros:

@Nicell: what work do you think is needed on the macros PR @Okke [banned]?

@okke-formsma:

At least some understanding of how the following could be done using the simple macros as a starting point:
- Different sets of keys being pressed on key down and key up of the macro
- having the ability to only trigger a “key press” or “key up” of a behaviour
- controlling the duration of pauses and key presses

I'm thinking along the lines of a `&keypress` and `&keyrelease` behavior, a `&delay` semi-behavior and something like `&wait-for-keyup`  as marker between the key-down and key-up sequences. `&delay` and `&wait-for-keyup` would only be meaningful within a macro definition

If the default delay between keypresses and the keypress duration is configurable at a macro level, it'd be easy to vary delays:
* want to keep A pressed for 100ms? `&keypress A &delay 100 &keyrelease A`
* want to press A, then tap B, release A on macro key up? `&keypress A &kp B &wait-for-keyup &keyrelease A`

Na-Cly, 2022-01-11:
Is there any plans to do on the fly recordable macros in the future?

okke-formsma, 2022-10-29:
We have a working implementation of macros right now, so let's put new feature requests (like recordable macros) in a new issue.

closed 2022-10-29 by okke-formsma