zmkfirmware/zmk #696: Bug: sticky key quick-release releases too quickly

okke-formsma, 2021-02-27:
My &sk LSFT key configured with `quick-release` doesn't do anything if I try to use the sticky behavior.  Pressing it triggers shift, but on key release the sticky behavior is stopped immediately.

azor738, 2021-02-28:
Can replicate. Closer inspection shows Shift being released right before the other key (~7 ms, think it's a limitation of the logging I'm using), no matter the value of release-after-ms

okke-formsma, 2021-04-05:
Should be fixed in that PR, @azor738 could you verify?

azor738, 2021-04-06:
> Should be fixed in that ***@***.*** could you verify?

Yes, but away from the keyboard until Sunday. Will check then! Thanks,

Hajime-Suzuki, 2021-04-06:
I also had this issue. For example, when I typed "test" quickly with `&sk LSHFT` without `quick-release`  I got "TEst" and "test" with it.  Now with this change, I get "Test" as expected at least with my nice!nano. 👍

PR #752 (2021-05-09): fix(behaviors): Fix sticky keys quick-release for normal keypresses
Quick release for sticky keys failed for non-layer keys. The sticky key
was released just before the key that was supposed to be modified was
handled.

The issue was caused by an error in the sticky key logic, which released
the sticky key before handling the key up event.

Fixes #696.
commit de4979bf (2021-05-09): fix(behaviors): Fix sticky keys quick-release for normal keypresses

Quick release for sticky keys failed for non-layer keys. The sticky key
was released just before the key that was supposed to be modified was
handled.

The issue was caused by an error in the sticky key logic, which released
the sticky key before handling the key up event.

Fixes #696.
closed 2021-05-09 by petejohanson