zmkfirmware/zmk #713: CMD & Shift Mod-Taps cause weird behaviors in correlation with combos.

ItsWaffIe, 2021-03-02:
Board: Nice!Nano 
Shield: CRKBD 
User Repo: https://github.com/ItsWaffIe/zmk-config

### Shift Issues 
Using shifted mod taps with keys used in combos cause the combo key to be spammed and if the combo key is shifted, caps lock becomes enabled but can't be turned off. This is fixed by resetting each half. In the video below, it was Shift + Y which is a combo on TY which outputs " : ". 
[Example](https://vimeo.com/user134859885/review/518652013/02adbc1930)

### CMD/Control Issues
I've tried multiple keycodes and all have resulted in this issue. The CMD keycode seems to get combined with whatever key I press next. In the video below, it was CMD + T.
[Example](https://vimeo.com/user134859885/review/518652962/de9e38969c)

okke-formsma, 2021-03-02:
[image]

your videos are still on private mode?

I don't really understand what you're exactly doing. Can you tell me what sequence of keys you press, and how they're configured?

e.g.
* press `&kp A`
* press combo on positions 1 and 2, which are mapped to `&kp X` and `&kp Y`
* release `&kp A`
* etc

ItsWaffIe, 2021-03-02:
> your videos are still on private mode?
Oops, fixed that 

> I don't really understand what you're exactly doing. Can you tell me what sequence of keys you press, and how they're configured?
Shift example: I have `&mt LSHFT A` and a combo that output `&kp COLON` when `&kp T` and `&kp Y` are pressed. If I hold `&mt LSHFT A` to get shift and press `&kp T` or `&kp Y`, it spams `&kp COLON` until I type other characters which are also now shifted and the only way to fix that is by resetting the controller. 

Command example: I have `&mt RCMD FSLH` and if I were to press any other key, like `&kp S` for example, Command & S remain held until the controller is reset.

okke-formsma, 2021-03-03:
I think this will be fixed by https://github.com/zmkfirmware/zmk/pull/697. Can you set your west.yml to point to my branch and flash that firmware to test?

ItsWaffIe, 2021-03-05:
Yep, seems to work nicely now. Thank you for your work.

closed 2021-03-05 by ItsWaffIe