zmkfirmware/zmk #715: Update Basic Setup information

schengnz, 2021-03-03:
This ticket is for collecting what needs to be updated on https://zmkfirmware.dev/docs/development/setup/ 

For Debian Zephyr SDK installation:

"The installation will prompt with several questions about installation location, and creating a default ~/.zephyrrc for you with various variables. The defaults should normally work as expected." is not longer correct. It does not ask questions nor generate `~/.zephyrrc`

schengnz, 2021-03-04:
(Here is a partial draft of a new page/section for Windows setup for ZMK development)

**Developing ZMK with Windows**
If you use a Windows based development machine, there are three main approaches to develop ZMK each with its own pros and cons.

  | Native Windows | Windows with Linux on WSL2 | Windows using Docker Container
-- | -- | -- | --
“Windows” look & feel | Yes | No, it is like having a Linux VM | Yes, mostly
Ability to compile native_posix build & integration tests | No - native_posix not supported on native Windows | Yes | Yes
Ability to interactive debug native_posix build & integration tests via VS Code | N/A | Yes | Yes
Ability to cross-compile device builds | Yes | Yes | Yes
Ability to interactive debug on-device via openocd/jagger etc | Yes | (May be) Right out of the box WSL2 Linux distributions has no ability to access USB hardware device (such as st-link or j-link). However it *may* be possible to recompile the Linux kernel to enable USB/IP support | Not locally - Docker container has no ability to access USB hardware device (such as st-link or j-link)
Simplicity of set up | Medium | Medium | Easiest
Instructions | See xxx | See xxx | See xxx
Other notes |   |   | Performance of compiling is much slower than the WSL2 approach

How to choose the three approaches:
- For users looking to compile builds for supported keyboards for a new keymap
-- Windows using Docker Container
- For core developers needing to interactively debug both integration test builds and device builds, a two-toolchains approach  may be necessary
-- Native Windows toolchain for on-device debugging
-- WSL2 toolchain for integration test debugging
- A fourth alternative is to install Linux Windows dual boot.

schengnz, 2021-03-04:
We should add to https://zmkfirmware.dev/docs/development/setup Basic Setup (other than for Docker approach) instructions to install gcc-multilib or similar.  If gcc-multilib is missing from the build environment, trying to compile integration tests or native_posix build will generate extremely obscure compiling issues. (Guess how I found out)

Installation instruction for gcc-multilib is currently not in Basic Setup, but somewhat obscurely in https://zmkfirmware.dev/docs/development/posix-board/

mcrosson, 2021-03-04:
this would be a good follow on after https://github.com/zmkfirmware/zmk/pull/437 is merged

closed 2025-07-31 by nmunnich