zmkfirmware/zmk #716: west test and integration test native_posix executables do not build correctly on Windows platform

schengnz, 2021-03-03:
`west test` appears to fail on Windows platform.
[image]

building integration test via `west build` also appears to fail.
[image]

schengnz, 2021-03-03:
The screenshots were provided by @joelspadin

joelspadin, 2021-03-05:
I'm not sure what file it couldn't find (or how to figure out which file it couldn't find), but I would not be surprised if some extra configuration is needed to point west to a valid Windows (not ARM cross-compile) toolchain.