zmkfirmware/zmk #718: Peripheral for split randomly disconnects and doesn't auto-reconnect.

Goldie643, 2021-03-04:
Board(s): Nice!Nano
Repo: https://github.com/Goldie643/zmk-config
Shield: mDactyl (Dactyl Manuform Mini)

At seemingly random intervals, approximately once every hour or so, the right hand of the board will disconnect and not provide any output. The left hand continues to work, but to re-enable the right hand the board needs to be re-paired by resetting both sides at the same time. Restarting the LH or RH separately doesn't fix the issue.

I enabled USB logging and kept it running in the background. I grabbed these logs right after it disconnected.

Left hand
```
[00:40:02.491,729] <dbg> zmk.split_central_disconnected: Disconnected: d6:39:f3:4d:8a:08 (random) (reason 8)
[00:40:02.492,065] <dbg> zmk.start_scan: Scanning successfully started
[00:40:02.492,309] <dbg> zmk.disconnected: Disconnected from d6:39:f3:4d:8a:08 (random) (reason 0x08)
[00:40:02.492,340] <dbg> zmk.update_advertising: advertising from 0 to 2
[00:40:02.492,340] <err> zmk: Advertising failed to start (err -69)
[00:40:02.556,121] <dbg> zmk.split_central_device_found: [DEVICE]: 4d:5c:dc:3f:79:0c (random), AD evt type 2, AD data len 28, RSSI -53
[00:40:02.635,131] <dbg> zmk.split_central_device_found: [DEVICE]: 6c:5b:74:58:4f:02 (random), AD evt type 3, AD data len 31, RSSI -42
[00:40:02.863,342] <dbg> zmk.split_central_device_found: [DEVICE]: 44:f9:07:4a:b8:e0 (random), AD evt type 0, AD data len 15, RSSI -49
[00:40:02.863,342] <dbg> zmk.split_central_eir_found: [AD]: 1 data_len 1
[00:40:02.863,372] <dbg> zmk.split_central_eir_found: [AD]: 2 data_len 2
[00:40:02.863,372] <dbg> zmk.split_central_eir_found: [AD]: 255 data_len 6
[00:40:03.062,133] <dbg> zmk.split_central_device_found: [DEVICE]: 51:51:51:b6:6f:93 (random), AD evt type 2, AD data len 28, RSSI -58
[00:40:03.827,117] <dbg> zmk.split_central_device_found: [DEVICE]: 53:51:a6:9e:45:03 (random), AD evt type 0, AD data len 17, RSSI -74
[00:40:03.827,148] <dbg> zmk.split_central_eir_found: [AD]: 1 data_len 1
[00:40:03.827,148] <dbg> zmk.split_central_eir_found: [AD]: 10 data_len 1
[00:40:03.827,148] <dbg> zmk.split_central_eir_found: [AD]: 255 data_len 9
[00:40:06.711,517] <dbg> zmk.split_central_device_found: [DEVICE]: 40:16:3b:f5:50:1f (public), AD evt type 3, AD data
```

Right hand
```
[00:40:00.297,607] <dbg> zmk.split_listener:
[00:40:02.485,229] <dbg> zmk.split_svc_pos_state_ccc: value 0
[00:40:02.486,877] <dbg> zmk.disconnected: Disconnected from c3:3d:aa:83:c7:4a (random) (reason 0x08)
[00:40:02.486,877] <dbg> zmk.disconnected: Active profile disconnected
[00:40:02.486,907] <dbg> zmk.update_advertising: advertising from 0 to 0
[00:40:02.486,938] <dbg> zmk.get_selected_endpoint: Only BLE is ready.
[00:40:02.486,968] <wrn> bt_att: Not connected
[00:40:02.486,999] <dbg> zmk.send_position_state_callback: Error notifying -57
[00:40:02.487,030] <dbg> zmk.send_position_state_callback: Error notifying -57
[00:40:02.487,060] <dbg> zmk.send_position_state_callback: Error notifying -57
[00:40:02.487,091] <dbg> zmk.send_position_state_callback: Error notifying -57
```

It's unclear to me which side is the issue, I thought the peripheral should automatically reconnect to the left after the disconnect but it appears that isn't the case (even just restarting the RH manually when previously connected requires a full re-pair).

DuMaM, 2021-11-07:
I got exactly the same issue on my dactly, but with particle xenon :/

mjrusso, 2021-11-08:
This might be fixed by #984 -- try giving that a shot.

DuMaM, 2021-11-08:
I'm not sure. I'm using recent version of zmk and issue is still present. 
Somebody suggested that issue maybe connected with bootloader.
Somehow I'm unable to find this post...
Anyway i will update a bootloader and if issue is still present then create an issue where all details will be listed.

Goldie643, 2021-11-10:
Sorry I never replied to this but I discovered the cause of the issue, but no solution. The cause was actually me keeping my phone in between the two keyboards. I have no idea why but it seems the phone's interference kicks off the peripheral, but moving the phone away doesn't help, you still need to re-pair. Sorry I don't have a more elegant solution!

catdevnull, 2022-09-25:
This has been happening to me too.

jceloi, 2023-10-30:
I can add it does happen for me too.