zmkfirmware/zmk #734: ZMK working when plugged on cable only

marcelotrevisani, 2021-03-21:
Hi folks,

First of all, thanks for the project! :)

I have a ``nice!nano`` board and I flashed a [simple code](https://github.com/marcelotrevisani/zmk-config/tree/master/config)  and it is working fine when plugged with USB cable

However, if I remove the cable and try to use the Bluetooth it is not working at all :/
I searched the documentation but I didn't see anything related to my case.
It is connecting fine using the Bluetooth, but the keys are not working for some reason?
Can you please give me some guidance here? Or maybe how can I get some logs to try to see what is happening?

I am using ``Ubuntu 20.10``
board: `nice!nano v1``
keymap: https://github.com/marcelotrevisani/zmk-config/blob/master/config/romac.keymap
It is a RoMac

joelspadin, 2021-03-21:
You might be able to get some useful info out of Ubuntu's bluetooth logs: https://wiki.ubuntu.com/DebuggingBluetooth

You can also try turning on logging on the keyboard to see if anything useful shows up there: https://zmkfirmware.dev/docs/development/usb-logging

If it is not pairing and connecting correctly, there should be some indication as to why in the logs. If it is connecting correctly, you'll also need to switch the preferred output to bluetooth, or it will keep trying to send to USB: https://zmkfirmware.dev/docs/behaviors/outputs

marcelotrevisani, 2021-03-21:
I checked and it is pairing fine. Is there a way to put the default to send the preferred output to be Bluetooth?

marcelotrevisani, 2021-03-21:
I also tried to add the ``&out OUT_TOG`` and it didn't work either :/

joelspadin, 2021-03-22:
Do the logs on the keyboard say the following? "Both endpoints are ready. Using 1" (0 is USB, 1 is bluetooth)

If the keyboard doesn't think that the bluetooth endpoint is ready, then it isn't pairing correctly.

marcelotrevisani, 2021-03-22:
I discovered that is something weird going on my Ubuntu machine, for some reason is not accepting the commands from the microcontroller. I found similar issues on the internet.
I tested on another computer with Windows and it worked fine. I am still trying to make it work, but the problem is not related directly to the zmk, if you want to you can close the issue then
Thanks for your attention! :)

joelspadin, 2021-03-22:
Weird. Well, if you do figure it out, feel free to put more details here in case anyone else runs into the same issue. Thanks!

closed 2021-03-22 by joelspadin