zmkfirmware/zmk #741: Support modifiers for C_VOL_UP and C_VOL_DN

jamesramsay, 2021-03-26:
When using MacOS, using Shift+Option+Volume Up/Down increases the volume by a 1/4 increments. This works with ZMK when they are separate keys chorded together.

However, it doesn't seem possible to combine into a single keypress with either:

```
&kp LS(LA(C_VOL_UP))
```

Or when using an encoder:

```
sensor-bindings = <&inc_dec_kp LS(LA(C_VOL_UP)) LS(LA(C_VOL_DN))>;
```

Are modifiers not supported for certain groups of key code?

okke-formsma, 2021-04-05:
It's supported for consumer keycodes, but not for sensor bindings.

jamesramsay, 2021-04-06:
@okke-formsma C_VOL_UP works just fine without modifiers. It also works fine when I hold modifier keys myself and turn the encoder.

It doesn't work when I apply the modifiers in the keymap `LS(LA(C_VOL_UP))`

caksoylar, 2022-04-03:
Would #1038 affect this behavior? It would be good to test with a fresh build.

petejohanson, 2022-04-03:
Yeah, re-testing would be good, I believe this is fixed by that PR.

caksoylar, 2022-04-06:
I tested with `LS(C_VOL_UP)` on Windows, below is the results of tapping `LS(INS)` and then the shifted volume up, both of which show the same sequence of mod+key presses:
[image]

I don't have a Mac to test the original keycode and see if it works, it would be good if someone can test and report.

dxmh, 2022-04-06:
Can confirm `&kp LS(LA(C_VOL_UP))` increases the volume in 1/4 increments for me.

* Latest version of zmk `main` 
* macOS 12.3
* `CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_BASIC=y` (in case it has relevance)

caksoylar, 2022-04-06:
Thanks @dxmh, sounds like this issue is fixed then.

closed 2022-04-06 by caksoylar

jamesramsay, 2022-04-10:
Thanks y'all 🙌