zmkfirmware/zmk #755: Docs Combo example does not compile

MvEerd, 2021-04-06:


okke-formsma, 2021-04-06:
Apparently the 'all layers' feature only works by not providing any value for the `layers` parameter, so it defaults to -1.

MvEerd, 2021-04-06:
The -1 is indeed superfluous in this case. I just wondered where the syntax error came from before offering a PR for the docs to say so.
Is it a devicetree limitation? I don't see a configuration affecting this apart from the default being set

okke-formsma, 2021-04-06:
https://docs.zephyrproject.org/latest/guides/dts/intro.html#syntax-and-structure 

> Devicetree nodes can also have properties. Properties are name/value pairs. Property values can be any sequence of bytes. In some cases, the values are an array of what are called cells. A cell is just a 32-bit unsigned integer.

MvEerd, 2021-04-06:
Ah, so they are indeed unsigned, that explains it. Thanks for pointing that part of the docs out!

Was indeed confused that you could still configure it as `-1` default

joelspadin, 2021-04-06:
> So it's actually kind of funky that the -1 default works :')

I'm pretty sure you're just getting `(uint32_t)(-1)` here, so the max possible uint32_t. Then since `combo_cfg::layers` is an array of signed integers, you get `(int8_t)(uint32_t)(4294967295)`, which fortunately happens to be -1.

MvEerd, 2021-04-15:
Fixed by #757

closed 2021-04-15 by MvEerd