zmkfirmware/zmk #756: Tapping capslock quickly doesn't work on OS X

okke-formsma, 2021-04-06:
Macs don't register capslock  if capslock is pressed and released too quickly.

also see https://github.com/tmk/tmk_keyboard/issues/390 / https://beta.docs.qmk.fm/developing-qmk/qmk-reference/config_options TAP_HOLD_CAPS_DELAY

As we don't want mac-specific delays in the codebase, we could fix this by adding a `tap-duration-ms` parameter to hold-taps so mac users can configure the key-up delay to 100ms so the capslock key is triggered properly. Also add an example in the docs with this specific example.

caksoylar, 2022-04-03:
One workaround that is now possible is to use a macro to trigger caps lock, where the tap duration can be customized. E.g. (not tested):
```devicetree
ZMK_MACRO(caps_macro,
    tap-ms = <200>;
    bindings = <&kp CAPS>;
)
```

RenatoMartinsFerreira, 2025-03-26:
> One workaround that is now possible is to use a macro to trigger caps lock, where the tap duration can be customized. E.g. (not tested):
> 
> ```
> ZMK_MACRO(caps_macro,
>     tap-ms = <200>;
>     bindings = <&kp CAPS>;
> )
> ```

Tested, worked flawlessly

caksoylar, 2025-04-18:
Apparently there is a way to fix this on macOS side, h/t @lesshonor for sharing: https://discussions.apple.com/thread/255801630