zmkfirmware/zmk #763: "<err> zmk: Security failed:" while pairing with MacOS

antonkril, 2021-04-15:
Controller: 2 Nice!Nano on Dactyl-Manuform build
MacOS version: 11.2.3 BigSur

Steps to reproduce:
1. Remove all connected devices from Bluetooth menu in MacOs
2. Reset Bluetooth module
3. remove `com.apple.Bluetooth.*.plist` files from `~/Library/Preferences` and `~/Library/Preferences/ByHost`
4. Reset both keyboard parts. Open Logs (notice the error messages)
Open Logs on the keyboard.
Logs after reset (notice the error messages present)
```
[tio 15:20:22] Connected
] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.279,205] <inf> fs_nvs: alloc wra: 0, c10
[00:00:00.279,205] <inf> fs_nvs: data wra: 0, 614
[00:00:00.287,353] <dbg> zmk.bvd_init: AIN2 setup returned 0
[00:00:00.289,245] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.290,283] <dbg> zmk.ble_profiles_handle_set: Setting BLE value profiles/4
[00:00:00.290,374] <dbg> zmk.ble_profiles_handle_set: Loaded 00:00:00:00:00:00 (public) address for profile 4
[00:00:00.290,771] <dbg> zmk.ble_profiles_handle_set: Setting BLE value active_profile
[00:00:00.292,572] <dbg> zmk.ble_profiles_handle_set: Setting BLE value peripheral_address
[00:00:00.294,586] <dbg> zmk.ble_profiles_handle_set: Setting BLE value profiles/0
[00:00:00.295,013] <dbg> zmk.ble_profiles_handle_set: Loaded 8c:85:90:d3:fc:02 (public) address for profile 0
[00:00:00.314,270] <dbg> zmk.zmk_ble_ready: ready? 0
[00:00:00.314,270] <dbg> zmk.update_advertising: advertising from 0 to 2
[00:00:00.315,246] <dbg> zmk.start_scan: Scanning successfully started
[00:00:00.315,856] <wrn> usb_cdc_acm: Device not configured or suspended, drop 1 bytes
[00:00:00.316,680] <inf> usb_hid: Device connected
[00:00:00.316,711] <dbg> zmk.get_selected_endpoint: Only USB is ready.
[00:00:00.316,711] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:00:00.316,741] <dbg> zmk.zmk_endpoints_send_report: usage page 0x0c
[00:00:00.316,741] <inf> zmk: Endpoint changed: 0
[00:00:00.316,802] <dbg> zmk.destination_connection: Address pointer 0x2000a6af
[00:00:00.316,802] <wrn> zmk: Not sending, no active address for current profile
[00:00:00.316,833] <dbg> zmk.destination_connection: Address pointer 0x2000a6af
[00:00:00.316,833] <wrn> zmk: Not sending, no active address for current profile
[00:00:00.319,793] <inf> usb_cdc_acm: Device suspended
[00:00:00.319,793] <inf> usb_hid: Device suspended
[00:00:00.319,824] <dbg> zmk.get_selected_endpoint: Only USB is ready.
[00:00:00.320,465] <dbg> zmk.bvd_sample_fetch: ADC raw 4072 ~ 2982 mV => 4183 mV
[00:00:00.320,465] <dbg> zmk.bvd_sample_fetch: Percent: 98
[00:00:00.320,465] <dbg> zmk.zmk_battery_update: Setting BAS GATT battery level to 98.
[00:00:00.321,258] <dbg> zmk.endpoints_handle_set: Setting endpoint value preferred
[00:00:00.321,441] <dbg> zmk.get_selected_endpoint: Only USB is ready.
[00:00:00.328,308] <dbg> zmk.kscan_gpio_init_0: Configured pin 9 on GPIO_0 for input
[00:00:00.328,338] <dbg> zmk.kscan_gpio_init_0: Configured pin 10 on GPIO_0 for input
[00:00:00.328,338] <dbg> zmk.kscan_gpio_init_0: Configured pin 11 on GPIO_1 for input
[00:00:00.328,369] <dbg> zmk.kscan_gpio_init_0: Configured pin 13 on GPIO_1 for input
[00:00:00.328,369] <dbg> zmk.kscan_gpio_init_0: Configured pin 15 on GPIO_1 for input
[00:00:00.328,369] <dbg> zmk.kscan_gpio_init_0: Configured pin 2 on GPIO_0 for input
[00:00:00.328,430] <inf> zmk: Welcome to ZMK!

[00:00:00.328,460] <dbg> zmk.kscan_gpio_configure_0: Configured GPIO 0
[00:00:00.390,441] <dbg> zmk.split_central_device_found: [DEVICE]: 8c:85:90:d3:fc:02 (public), AD evt type 0, AD data len 14, RSSI -47
[00:00:00.390,441] <dbg> zmk.split_central_eir_found: [AD]: 1 data_len 1
[00:00:00.390,441] <dbg> zmk.split_central_eir_found: [AD]: 255 data_len 9
[00:00:00.417,663] <inf> usb_cdc_acm: Device resumed
[00:00:00.417,694] <inf> usb_cdc_acm: from suspend
[00:00:00.417,694] <inf> usb_hid: Device resumed
[00:00:00.417,694] <inf> usb_hid: from suspend
[00:00:00.417,724] <dbg> zmk.get_selected_endpoint: Only USB is ready.
[00:00:00.417,785] <inf> usb_hid: Device reset detected
[00:00:00.417,816] <dbg> zmk.get_selected_endpoint: No endpoints are ready.
[00:00:00.417,816] <dbg> zmk.zmk_endpoints_send_report: usage page 0x07
[00:00:00.417,816] <err> zmk: FAILED TO SEND OVER USB: -19
[00:00:00.417,816] <dbg> zmk.zmk_endpoints_send_report: usage page 0x0c
[00:00:00.417,846] <err> zmk: FAILED TO SEND OVER USB: -19
[00:00:00.417,846] <inf> zmk: Endpoint changed: 1
[00:00:00.437,652] <dbg> zmk.split_central_device_found: [DEVICE]: 46:88:0f:13:65:ed (random), AD evt type 2, AD data len 28, RSSI -70
[00:00:00.523,590] <dbg> zmk.split_central_device_found: [DEVICE]: 64:16:66:48:e4:da (random), AD evt type 0, AD data len 31, RSSI -77
[... 46 lines cut ...]
[00:00:03.872,894] <dbg> zmk.split_central_discovery_func: [ATTRIBUTE] handle 45
[00:00:03.872,955] <dbg> zmk.split_central_subscribe: [SUBSCRIBED]
```
5. Confirm that both parts are connected and work through USB
6. Open Bluetooth Preference dialog in MacOs
7. Find the Advertised keyboard device and click "Connect"

Expected result: the keyboard connects and sends signals to MacOs
Actual result: the connection process either hangs or the device is shown as connected but does not communicate with MacOs

Pairing logs (notice error messages):
```
[00:00:49.336,730] <dbg> zmk.split_central_connected: Connected: 8c:85:90:d3:fc:02 (public)
[00:00:49.336,730] <dbg> zmk.split_central_process_connection: Current security for connection: 1
[00:00:49.338,104] <dbg> zmk.split_central_process_connection: New connection params: Interval: 12, Latency: 0, PHY: 1
[00:00:49.338,165] <dbg> zmk.connected: Connected thread: 0x20003ae8
[00:00:49.338,195] <dbg> zmk.connected: Connected 8c:85:90:d3:fc:02 (public)
[00:00:49.338,195] <dbg> zmk.update_advertising: advertising from 0 to 2
[00:00:49.374,206] <err> zmk: Security failed: 8c:85:90:d3:fc:02 (public) level 1 err 4
[00:00:50.153,778] <inf> bas: BAS Notifications enabled
[00:00:50.290,130] <err> zmk: Security failed: 8c:85:90:d3:fc:02 (public) level 1 err 4
[00:00:50.350,219] <err> zmk: Security failed: 8c:85:90:d3:fc:02 (public) level 1 err 4
```

**Note**: the device works well with Iphone

antonkril, 2021-04-16:
I was able to connect the keyboard after making Bluetooth discoverable with `blueutil --discoverable 1`. But can not confirm that it was the reason.

CelsoSantos, 2021-05-09:
I am also facing this exact issue...

kh411d, 2021-10-04:
Controller: nrfmicro 1.4
Board: Contra
I can't seem to get the Bluetooth paired on Mojave either,

```
default	08:55:52.203194 +0700	bluetoothd	Read Device Name: cfc23cd0
default	08:55:52.221247 +0700	cloudpaird	Setting HeySiri disabled for Contra Keyboard
default	08:55:52.622994 +0700	bluetoothd	Updated HID Setup Status (Read) 36 <GATTCharacteristic:0x7fedcff36420 handle:0x0024 valueHandle:0x0025 uuid:0x2A4D properties:0x12 ( Read Notify )>
default	08:55:52.652920 +0700	bluetoothd	Updated HID Setup Status (Read) 32 <GATTCharacteristic:0x7fedcff31b10 handle:0x0020 valueHandle:0x0021 uuid:0x2A4D properties:0x12 ( Read Notify )>
default	08:55:52.653107 +0700	bluetoothd	Attempting To Create HID For Mapping {(
    2D4C9C66-257A-4CEB-8DCB-A9F2EE420701
)}
error	08:55:52.653259 +0700	bluetoothd	Cannot create HID device, link is not encrypted
default	08:55:52.682893 +0700	bluetoothd	***** [handleATTErrorResponse].
error	08:55:52.683445 +0700	bluetoothd	***** [handleATTErrorResponse] - Insufficient Encryption.
default	08:55:52.683611 +0700	bluetoothd	***** [handleATTErrorResponse] - Link is not encrypted, perform pairing with MITM disbled.
default	08:55:52.683676 +0700	bluetoothd	***** [startLESecurityManagerPairingSecureConnection] Start secure pairing with mitm: 0, bonding: 1, sc: 1, keypress: 0
default	08:55:52.684287 +0700	bluetoothd	***** [startLESecurityManagerPairingSecureConnection] - Master: _localAuthReq: 0x9, localInitiatorKeyDistribution: 0x3, localResponderKeyDistribution: 0x3
default	08:55:52.684341 +0700	bluetoothd	***** [pairingRequest] Send Pairing Request
default	08:55:52.899863 +0700	bluetoothd	***** [l2capChannelData] - Received Pairing Failed.
error	08:55:52.899922 +0700	bluetoothd	*****[handlePairingFailed] reason = 3.
default	08:55:52.900070 +0700	bluetoothd	***** lePairingCompleted for deviceAddr = e2-13-9f-05-33-77 (name = Contra Keyboard) with status = e00002bc
error	08:55:52.900215 +0700	bluetoothd	Unable to Subscribe to HID Report <GATTDescriptor:0x7fedcff024f0 handle:0x0026 uuid:0x2902> - Destroying HID If Created
```

CelsoSantos, 2021-12-21:
@Notaduck IIRC, my issue was that I already had a profile paired to that machine. 

Try iterating over the 5 (I think?) profiles and reset each one. Then select the profile you want to use and pair back. Don't forget to also tell the machine to forget the keyboard otherwise, it still won't work)

Notaduck, 2021-12-22:
> @Notaduck IIRC, my issue was that I already had a profile paired to that machine and was attempting to re-pair on another profile.
> 
> Try iterating over the 5 (I think?) profiles and reset each one. Then select the profile you want to use and pair back. Don't forget to also tell the machine to forget the keyboard, otherwise, it still won't work.

Thanks for the feedback, i used the setting reset file and it worked afterwards :)

CelsoSantos, 2021-12-22:
No worries :). Serves as a future reference for those that may find the same issue.

Anyway, I'd say this is user error, not zmk related. I believe this issue might be closed, in the meantime.

antonkril, 2022-01-03:
My issue was that the keyboard lost connection to MacOS and could not reconnect.

But it seems to have been fixed in the latest release*

closed 2022-01-03 by antonkril

nosarthur, 2023-06-06:
> @Notaduck IIRC, my issue was that I already had a profile paired to that machine and was attempting to re-pair on another profile.
>
> Try iterating over the 5 (I think?) profiles and reset each one. [...]

Thanks. This iterating and resetting finally fixed my connection

shuji-koike, 2025-02-05:
https://zmk.dev/docs/troubleshooting/connection-issues#building-a-reset-firmware
`settings_reset` firmware worked for me.