zmkfirmware/zmk #767: Feature request: Select previously active Bluetooth profile

dxmh, 2021-04-21:
It would be great to be able switch to the previously active Bluetooth profile. 

For example — I’m typing away on profile 1, then I manually switch to profile 2 for a small task. I’d then like to get back to profile 1 using something like this:

```
&bt BT_SEL {{ previously active profile }}
``` 

This functionality would make the experience of switching back and forth between profiles similar to `ctrl`+`tab` window switching. 

This could be useful also in macros, for example to pause the media on another host and return back to the current host:

```
macro_toggle_media: macro_toggle_media {
  compatible = "zmk,behavior-macro";
  label = "macro_toggle_media";
  #binding-cells = <0>;
  bindings = <&bt BT_SEL 1>, <&kp C_PLAY_PAUSE>, <&bt BT_SEL {{ previously active profile }}>;
};

```

auipga, 2021-11-01:
I like the idea. **Prev** and also **next** (some sort of history) would be awesome.

Dimentium, 2024-04-24:
I've added PR #2278
It works as expected. You can use index 255 to switch back to the previously used profile.