zmkfirmware/zmk #775: kscan-gpio-matrix debouncing is broken in polling mode

atx, 2021-04-29:
In polling mode, the `kscan-gpio-matrix` [schedules a 10ms timer](https://github.com/zmkfirmware/zmk/blob/main/app/drivers/kscan/kscan_gpio_matrix.c#L224), which [submits a matrix scan job](https://github.com/zmkfirmware/zmk/blob/main/app/drivers/kscan/kscan_gpio_matrix.c#L237) to the shared system queue. There is no additional debouncing algorithm, so the scan [produces](https://github.com/zmkfirmware/zmk/blob/main/app/drivers/kscan/kscan_gpio_matrix.c#L178) a key change notification every time a key state changes between runs.

### Issues 
The largest problem is that since the system work queue is shared between system components, a scanning job may get delayed by an arbitrary amount of time. This means two scanning jobs may get executed right after one another. If a key press occurs at an unfortunate moment, the switch bouncing may cause multiple key presses to get registered.

One way of observing this problem is by having an oscilloscope trigger on a pulse shorter than 10ms on one of the matrix driver pins, as illustrated (ignore the cursors) by: 

[image]

Another minor issue is latency. Since the polling interval is also used for debouncing, there is up to a 10ms delay before a key press gets registered.

### Suggested solution

This can be solved by keeping a last-changed timestamp for each key in the matrix. Since we have an absolute timestamp, the fact that the scans may sometimes get delayed is no longer an issue. This also solves the latency issue, as we can separate the debounce interval and the poll rate, polling significantly faster.

Running the scanning jobs in a work queue/thread with higher priority than the main thread may also be a good idea, as it allows harder real time latency guarantee. Not sure if its worth the effort though...

A stream-of-conciousness implementation of the above is [here](https://github.com/atx/zmk/blob/polling/app/drivers/kscan/kscan_gpio_matrix_polled.c). The main thing missing (together with a lot of minor things) at the moment is interrupt support, as I do not have hardware which supports level-triggered interrupts to test it on. If there is interest, I can definitely get it to a mergeable state, but the interrupt support would have to be implemented/tested by someone else.   

### Notes
This was observed on an STM32F405 running at 120MHz, so is unlikely to be caused by a lack of processing power. The debugging CDC ACM port was enabled, which might have contributed to the frequency of the issue. 

I am not sure how often this happens in interrupt mode. Since the key scan job gets scheduled to run after at least 5ms passed since the interrupt registered a key press, it should not be an issue, at least as long as no key is held continuously.

cdc-mkb, 2021-09-13:
I observed debouncing issues in polling mode too :(

joelspadin, 2021-09-14:
#888 provides an improved debouncing algorithm.

petejohanson, 2022-05-26:
Can you please try with the improved debounce changes, which are now included in both the matrix and direct kscan drivers?