zmkfirmware/zmk #787: &sk RC(LEFT_SHIFT) doesn't work

okke-formsma, 2021-05-09:
Modified keycodes or combined modifiers can't be used as sticky keys right now. They are released too early, because the sticky key doesn't recognise it's own keypress.

This will be fixed once https://github.com/zmkfirmware/zmk/pull/670 lands.

Discussed on discord https://discord.com/channels/719497620560543766/719909884769992755/840966539951079454

okke-formsma, 2021-05-09:
```
#include <dt-bindings/zmk/keys.h>
#include <behaviors.dtsi>
#include <dt-bindings/zmk/kscan_mock.h>

/ {
	keymap {
		compatible = "zmk,keymap";
		label ="Default keymap";

		default_layer {
			bindings = <
				&sk RC(LEFT_SHIFT) &sk LEFT_CONTROL
				&kp A &none
			>;
		};
	};
};

&kscan {
	events = <
		ZMK_MOCK_PRESS(0,0,10) 
		ZMK_MOCK_RELEASE(0,0,10) 
		ZMK_MOCK_PRESS(1,0,10) 
		ZMK_MOCK_RELEASE(1,0,1200)
	>;
};

pressed: usage_page 0x07 keycode 0xe1 implicit_mods 0x00 explicit_mods 0x10
released: usage_page 0x07 keycode 0xe1 implicit_mods 0x00 explicit_mods 0x10
pressed: usage_page 0x07 keycode 0x04 implicit_mods 0x00 explicit_mods 0x00
released: usage_page 0x07 keycode 0x04 implicit_mods 0x00 explicit_mods 0x00

[00:00:00.040,000] <dbg> zmk.on_sticky_key_binding_released: Another key was pressed while the sticky key was pressed. Act like a normal key.
```

caksoylar, 2022-04-03:
Is this also fixed by #1122, can anybody confirm?

nguyendown, 2022-12-17:
I can't reproduce the issue. It is likely that it has already been fixed.
```
pessed: usage_page 0x07 keycode 0xE1 implicit_mods 0x00 explicit_mods 0x10
pressed: usage_page 0x07 keycode 0x04 implicit_mods 0x00 explicit_mods 0x00
released: usage_page 0x07 keycode 0xE1 implicit_mods 0x00 explicit_mods 0x10
released: usage_page 0x07 keycode 0x04 implicit_mods 0x00 explicit_mods 0x00
```

closed 2022-12-17 by caksoylar