zmkfirmware/zmk #793: Improve structure and naming of docs

okke-formsma, 2021-05-13:
I've been thinking about improving the naming and structure of the docs.

I don't  really like the name "behaviors" for end users, maybe using "keys" in the section title will point people to the right direction sooner.

https://docs.google.com/spreadsheets/d/1ABY0sRLw2bSuuwUTekm6ScSCuk0n35nU-C5-ilYgkfg/edit?usp=sharing

Changes:
* split behaviors into "Basic Keys" and "System Keys"
* rename "Codes" to "Key Codes"
* rename "Miscellaneous" to "None & Trans"
* move "Beta Testing" from "Features" to "Getting Started"

dxmh, 2021-05-13:
As a relatively new ZMK end user, it did take me a while to get familiar with current structure. Only after I had read the sections on features, behaviours and key codes did I have a clear understanding of the differences between each. So I think what you’re proposing is a great idea.

Initially when I was checking out the documentation, I looked under _Features_ for things like mod-tap and combos (perhaps because of the [“Features” table on the homepage of the docs](https://zmkfirmware.dev/docs/)?) — whereas of course these things are under _Behaviours_. 

> split behaviors into "Basic Keys" and "System Keys"

I like that with this approach everything is a _key_ as far as the end user is concerned, although I wonder if the word “keys” could result in users confusing these pages with key codes (keyboard _output_), whereas behaviours are more keyboard _functionality_?

Also, if restructuring the documentation, it could be worth keeping the comments in this PR in mind too: #722 🙂

joelspadin, 2021-05-14:
The main issue I see with renaming "behaviors" to something else is that the underlying implementation still calls them behaviors, so if you are developing the code, or if you just want to customize hold taps or create macros, now you have to learn two words for the same thing.

What if we simply renamed "behaviors" to "key behaviors" to make it clearer that it means "what happens when you press a key"?

okke-formsma, 2022-10-29:
a lot of effort has been put into docs in the last year, I think we can close this :)

closed 2022-10-29 by okke-formsma