zmkfirmware/zmk #805: Bluetooth security failing on windows machine

aureumleonis, 2021-05-24:
I've been using a split/nice!nano handwired built for a few months already with no issues on my personal PC. On the other hand, The keyboard pairs correctly with the my work laptop but doesn't send any keypresses. Both of these are windows 10 machines. I finally took the initiative to enable debugging and saw this happen constantly (a couple of times per second)

```
[00:25:58.514,068] <err> zmk: Security failed: <redacted> (public) level 1 err 2
[00:25:58.639,770] <dbg> zmk.split_central_disconnected: Disconnected: <redacted> (public) (reason 19)
[00:25:58.639,831] <dbg> zmk.disconnected: Disconnected from <redacted> (public) (reason 0x13)
[00:25:58.639,862] <dbg> zmk.update_advertising: advertising from 2 to 2
[00:25:58.678,680] <dbg> zmk.split_central_connected: Connected: <redacted> (public)
[00:25:58.678,680] <dbg> zmk.split_central_process_connection: Current security for connection: 1
[00:25:58.679,901] <dbg> zmk.split_central_process_connection: New connection params: Interval: 48, Latency: 0, PHY: 1
[00:25:58.679,992] <dbg> zmk.connected: Connected thread: 0x20002e40
[00:25:58.679,992] <dbg> zmk.connected: Connected <redacted> (public)
[00:25:58.680,023] <dbg> zmk.update_advertising: advertising from 0 to 2
```

I may have the beginning/end of the log wrong but this is still the part that repeats. Does anybody know what the problem could be? Is there anything else I can I do to find more data about the problem?

haywirephoenix, 2021-06-02:
Have you tried switching the bluetooth profile for your work machine? 
It's right here in the docs https://zmkfirmware.dev/docs/behaviors/bluetooth/#bluetooth-pairing-and-profiles

aureumleonis, 2021-06-03:
I've tried these two scenarios

Personal + work PCs
* Pair personal PCs on profile 0 (keyboard pairs and connects)
* Pair work PC on profile 1 (keyboard only pairs, shows the logs in the description)

Only work PC
* Clear all profiles
* Pair work PC on profile 0 (keyboard only pairs, shows the logs in the description)

haywirephoenix, 2021-06-03:
Maybe related: I've noticed an issue when using BT_CLR you have to remove+repair the paired device from the list and restart the pc before it will register input. 

It may be worth doing a full settings clear and trying again without touching BT_CLR
https://zmkfirmware.dev/docs/troubleshooting/#split-keyboard-halves-unable-to-pair

I'll see if it's the same for switching profiles

aureumleonis, 2021-06-03:
This is not a new problem BTW. I've dealt with it for around 5 months. I've done all the permutations of resets (BT_CLR and flashing), restarts and even tried a separate bluetooth dongle. No luck.

The other thing I've noticed though is some of the time the board will just not pair at all after clearing. I get the message "Couldn't connect. Try connecting again."

Kikokeyco, 2021-06-17:
I'm having the same problem with my nrf52840 based custom board, it will connect, allow me to type a couple inputs, and then disconnects, on both my mac and my iphone.

aureumleonis, 2021-11-01:
I just rebuilt and flashed firmware with latest ZMK hoping that maybe the zephyr 2.5 thing fixed the issue but it didn't. Is there something else I can do to further debug this problem? It's kind of a bummer that I have this wireless keyboard that I can't use wirelessly... I was considering building another one but I'm having doubts since it's likely going to have the same problem.

aureumleonis, 2021-11-03:
I just tested my board with BlueMicro and that firmware does connect to the work laptop successfully.

KostkaBrukowa, 2021-12-01:
Has anyone find a way to fix this? I have the same problem and I also tried BlueMicro and it works fine

Avada-sao, 2022-09-25:
I got the same problem.On Windows it can’t detect any input.But on my Ubuntu,and my IOS,it works.Also,I tested my keyboard with BlueMicro on my Windows,it works, too .

sadekbaroudi, 2022-12-19:
Just to add to the list here, I have experienced the same thing. I have tested two different keyboards on 6 different devices. 

Consistent behavior between Keyboard A and Keyboard B (both nice!nanos). All machines are Windows or Android OS:
* Laptop 1 (personal, windows 10) - Failed
* Laptop 2 (work, windows 11) - Success
* Laptop 3 (personal / kids, windows 10) - Success
* Desktop (personal, windows 11) - Failed
* Phone - Success
* Tablet - Success

Similar to comments above, I have tried clearing profiles, flashing the reset settings uf2, clearing and re-pairing. I even tried connecting after a full reformat and reinstallation of windows on the personal laptop 1. Consistent behavior.

caksoylar, 2022-12-19:
@sadekbaroudi Are you sure your BT adapters support BLE? It needs to support BT version 4.2 for the security ZMK uses, if I remember correctly.

sadekbaroudi, 2022-12-20:
Thanks for confirming that!

greyhatmiddleman, 2022-12-27:
I think this might be a zmk firmware issue with BT5.2 or greater. I have another nicenano with kmk (ble enabled) and I can connect, remove, clear and re-pair for the same host.  Where as with ZMK, I'm unable to repair entirely.

As for other systems with BT5.0, zmk works accordingly.

greyhatmiddleman, 2023-01-01:
> I think this might be a zmk firmware issue with BT5.2 or greater. I have another nicenano with kmk (ble enabled) and I can connect, remove, clear and re-pair for the same host. Where as with ZMK, I'm unable to repair entirely.
> 
> As for other systems with BT5.0, zmk works accordingly.

It seems that Pete's branch [zephyr/3.1](git clone --branch zephyr/3.1-upgrade https://github.com/petejohanson/zmk.git) might be a good fix. I have a corne (simple, no oled or rgb) with nice_nanos and so far I'm able to clear the profile and remove the device from windows and re-pair the keyboard. One thing to note is that once you remove the device from windows you need to restart the host, then you can re-pair the keyboard.

I'm assuming zephyr 3.1[3.2 based on app/west.yml] seems to improve bluetooth connectivity with hosts.

caksoylar, 2023-01-02:
For reference, #1499 is the associated PR for the Zephyr upgrade.

greyhatmiddleman, 2023-01-02:
> For reference, #1499 is the associated PR for the Zephyr upgrade.

Awesome, looking forward to the merge. If I run into any issues with the PR branch, I report them accordingly.

sadekbaroudi, 2023-01-03:
Oooh, yay! Looking forward to trying that out as well!

vanessa-luna, 2023-01-31:
I also really hope this gets resolved. My new Corne with niceNano are not staying reliably connected to my Win10 machine. I increased the bluetooth power, and removed all profiles and started over to no avail. IT will work temporarily then stop until I remove and re-pair.

My laptop has a Intel® Wi-Fi 6E AX210 No vPro®... My laptop reports LMP 12.14198

greyhatmiddleman, 2023-02-02:
> I also really hope this gets resolved. My new Corne with niceNano are not staying reliably connected to my Win10 machine. I increased the bluetooth power, and removed all profiles and started over [...]

If you don't have too much features with your corne keyboard, you can use the following repo and branch:
https://github.com/petejohanson/zmk/tree/zephyr/3.1-upgrade

I've been running this for over a month and I don't experience any connectivity issues.

vanessa-luna, 2023-02-02:
I hate to ask this kind of question here @greyhatmiddleman but  I am out of practice. With this branch I will have to run and compile on my local machine, and I can follow the ZMK docs to get that going yes? Also why if I don't have too many features? Is this branch lacking in features? Thank you for the info.

caksoylar, 2023-02-02:
> With this branch I will have to run and compile on my local machine, and I can follow the ZMK docs to get that going yes?

You can do that following [ZMK development docs](https://zmk.dev/docs/development/setup), or [change which ZMK fork/branch your config repo points to](https://zmk.dev/docs/features/beta-testing) to build using Github actions.

> Also why if I don't have too many features? Is this branch lacking in features? Thank you for the info.

This branch is related to PR #1499, features like underglow/displays might have issues potentially, but I believe it is pretty much ready for final testing so you shouldn't expect many issues.

greyhatmiddleman, 2023-02-03:
@vanessa-luna @caksoylar beat me to answer your questions. He is correct with his response.

zhenchaopro, 2023-02-06:
It seems that the both master and PR #1499 don't work with BT 5.2( Window 10 with LMP 11).  My Macbook Pro works fine.

vanessa-luna, 2023-02-08:
> It seems that both master and PR #1499 don't work with BT 5.2( Window 10 with LMP 11). My Macbook Pro works fine.

The PR has worked for me in Win10 with LMP 12. Have you also ensured you remove ALL BT profiles on the keyboard before pairing after the firmware update? It will be discouraging if this doesn't fix it for everyone. When you say it doesn't work, does it fail in the same way others are describing above?

zhenchaopro, 2023-02-10:
@vanessa-luna Yeah, I tried it for several times. The main branch works at the first time, it fails to connect after restart. PR https://github.com/zmkfirmware/zmk/pull/1499 fails at the first time.

greyhatmiddleman, 2023-02-11:
> @vanessa-luna Yeah, I tried it for several times. and the [Split battery reporting](https://github.com/zmkfirmware/zmk/pull/1243) branch works at the first time, it fails to connect after restart. PR #1499 fails at the first time.

When I reflash my controllers, I normally would initially flash with a 'settings_reset' uf2 firmware to clear anything in the controllers before flashing any new firmware. Check [Troubleshooting](https://zmk.dev/docs/troubleshooting#option-1-build-reset-uf2-in-zmk-config), although this is in reference to the halves not pairing, it ensures anything stuck in Flash is cleared.

Once you've created a 'settings_reset' uf2 firmware you can keep this for future resets, no need to constantly recompile a settings reset uf2.

j-ea, 2023-05-22:
It looks like PR #1499 was merged, but unfortunately the Windows BLE 5.1+ issue still persists.

My nice!nano board does not connect on Windows 10 w/ BLE 5.1. I get the error "Couldn't connect. Try connecting again."

It works when the connected via USB.
It works on the same machine while booted into different flavors of Linux. 
It also works on a different Windows 10 device w/ BLE 4.2.

caksoylar, 2023-05-22:
Dual booting will always have issues since hardware keys are shared across different OS: #549. This also sounds consistent with the error message you are getting.

j-ea, 2023-05-22:
@caksoylar, thank you for the quick reply! I'll give this a try tonight and reply here.

j-ea, 2023-06-01:
This was correct. Thanks again. Dual booting was causing the issues.

I'm dual booting with Linux and Windows. I see the PR was never merged. I'm happy to update it, add relevant Linux info, and submit another PR if anyone would find it useful.

caksoylar, 2023-06-01:
That would be good but I think it would be better if we directly refer to an existing resource rather than maintaining this ourselves. [Arch wiki BT page](https://wiki.archlinux.org/title/Bluetooth) (section 2.1) is a good and well-maintained candidate, although it doesn't include information on the Windows-macOS pairing.(naturally).

l3ztum, 2023-06-07:
I'm having the same problem. 

My xiao ble board does not connect on Windows 10 Enterprise with integrated BLE 5.2. Settings show toggle between paired and connected.

What works:
- connected via USB (even when only master is connected, so communication between halfs is ok)
- works on other PCs with 4.2 adapter, linux and windows 
- works on Android (P20Pro) also 4.2

What i tried:
- resetting with settings-reset flash
- reflashing firmware
- deleting profiles with BT_CLR

sheedapistawl, 2023-07-22:
I have the same problem, and Zephyr 3.4 test did not solve the issue either.

jacksonSingleton, 2023-09-26:
Having this problem on Windows, I can pair to MacOS just fine. But Windows gives an error: "try connecting your device again" when using a cleared profile. I'm using nice!nanos, no dual boot, and am able to successfully pair other BT devices such as wireless game controllers to the same machine. Any solutions at all?

alternateved, 2023-09-27:
I think I might be having the same issue on Fedora Linux with laptop with BT 5.3:
```
[00:09:24.707,855] <dbg> zmk: split_central_connected: SKIPPING FOR ROLE 1
[00:09:24.707,885] <dbg> zmk: connected: Connected thread: 0x20004dd0
[00:09:24.707,977] <dbg> zmk: connected: Connected 7C:B5:66:65:9D:97 (public)
[00:09:24.709,289] <dbg> zmk: update_advertising: advertising from 0 to 2
[00:09:24.768,157] <dbg> zmk: auth_pairing_accept: role 1, open? yes
[00:09:24.908,599] <err> zmk: Security failed: 7C:B5:66:65:9D:97 (public) level 1 err 9
[00:09:24.914,489] <dbg> zmk: split_central_disconnected: Disconnected: 7C:B5:66:65:9D:97 (public) (reason 42)
[00:09:24.914,581] <dbg> zmk: disconnected: Disconnected from 7C:B5:66:65:9D:97 (public) (reason 0x2a)
[00:09:24.914,611] <dbg> zmk: update_advertising: advertising from 2 to 2
```

No combination of clearing Bluetooth profiles, resetting firmware, reflashing helped. Once I tried branch from https://github.com/zmkfirmware/zmk/pull/1853 I was finally able to connect and use the keyboard.

gabeklavans, 2023-10-11:
I believe I'm facing the same issue. Haven't got USB debugging yet but I'm seeing:

- It pairs and works perfectly with my Windows desktop using a motherboard from a few years ago
- It fails to pair on my newer Windows laptop (with the error in the original post)
- No amount of settings resets or profile clearing fixes the behavior on the laptop, but the desktop remains fine throughout

I might try #1853 but it looks pretty WIP right now

NickPancakes, 2023-10-13:
Another person who suddenly ran into this issue yesterday, presumably (I didn't capture any debug logs). 
I only connect via BT to a single Windows 11 PC, but I also often connect wired to a KVM that switches between the Win11 PC and and an M1 Macbook.
Tried: Resetting BT profiles, rebuildling+reflashing zmk, un/reinstalling BT devices+drivers in Win11.
However, the Zephyr 3.4 branch linked above (and modified with the diffs from the latest comments on the PR) appears to have fixed the issue for me, for now at least.

- Windows 11
- Motherboard integrated BT 5.0
- 2x Nice!Nano v2
- Split shields (keeb.io Viterbi)

joelspadin, 2023-11-03:
For anyone who started getting this in the last couple of months, check the version of your Bluetooth driver (Settings > Apps > Installed Apps > search for "bluetooth"). If you have Intel Wireless Bluetooth driver version 22.250.0, try uninstalling it and [installing version 22.240.0 instead](https://www.intel.com/content/www/us/en/download/18649/784735/intel-wireless-bluetooth-for-windows-10-and-windows-11.html).

fulaphex, 2023-11-03:
for the problems with pairing with Windows 11, I found https://github.com/zmkfirmware/zmk/pull/1853 to be quite helpful. As it stands, the PR is very much work-in-progress, as of writing this message I had to revert to https://github.com/zmkfirmware/zmk/pull/1853/commits/540c6f6e1b7281c9f1c36535667fec02fd3a5669 to even be able to compile zmk, but after building the firmware with the 3.4, windows manages to pair. I still cannot get the pairing to work consistently 100%, but once it's paired - it connects without any issues and that's all I care for right now

jmding8, 2023-11-15:
I just ran into this issue. I tried:
1. clear all the profiles from both the board
2. unpair from the PC
3. restart the PC
4. reset the board
What ultimately fixed it for me was doing a settings reset (https://zmk.dev/docs/troubleshooting#split-keyboard-halves-unable-to-pair).

TomzBench, 2023-11-24:
> > For anyone who started getting this in the last couple of months, check the version of your Bluetooth driver (Settings > Apps > Installed Apps > search for "bluetooth"). If you have Intel Wireless Bluetooth driver version 22.250.0, try uninstalling it and [installing version 22.240.0 instead](https://www.intel.com/content/www/us/en/download/18649/784735/intel-wireless-bluetooth-for-windows-10-and-windows-11.html).
> 
> I used the Roll Back Driver feature in my Win11 device manager, which put it back at 22.230.0.2. I can confirm with this driver, my ZMK corne works over bluetooth on my machine now.
> 
> Though I'm not sure how I feel on recommending that people downgrade crucial drivers as a fix... but it does work for now.

Can confirm this solution worked for me as well! (And I couldnt give a shit about the security of my windows shitbox. its a fine solution. Though I'm curious if this is something that needs to be patched on the keyboard side? or a bug in intel driver that can eventually be patched in a future update)

phuertay, 2023-12-21:
Solved by following #1487. I'm using a custom split w/Nice!nanos on Windows 11 with the newest Intel driver (22.250.0.2)

1. Add "CONFIG_BT_CTLR_PHY_2M=n" to the .conf file (in my case, zmk-config/config/dacman56.conf)
2. Load settings_reset.uf2 on both sides
3. Load firmware on both sides
4. Re-paired

jakubgrygiel, 2024-01-13:
> For anyone who started getting this in the last couple of months, check the version of your Bluetooth driver (Settings > Apps > Installed Apps > search for "bluetooth"). If you have Intel Wireless [...]

I had the same issue (corne split, 2x nice nanos). I recently updated my Windows drivers and returning to an older bluetooth version solved the problem

gabeklavans, 2024-02-26:
Looks like the latest release that includes Zephyr 3.5 fixes this issue. I've rebuilt my firmware using the docker container at the main tag, and I am having no issues connecting when my Intel Bluetooth driver is updated to 22.250.0.2

sharifrazzaque, 2024-03-23:
Same here. My keyboard (running ZMK from July 2023) would not pair with my new Windows 11 laptop. I Rebuilt with 3.5 just now and it paired to the same win 11 laptop w/o any problems. I did not have to reset keyboard settings, roll back Bluetooth drivers or anything else besides update to ZMK 3.5

rayanpiro, 2024-03-31:
Same here. Using win 11 and a BT 5.3 dongle. If I try to connect with my laptop's intel card it works properly.

maslomeister, 2024-05-27:
Zephyr 3.5 Corne, NRF52840, Win 11 23H2, Generic bluetooth adapter 5.3 (tried 2 different ones)

I am able to pair my keyboard fine and use it, but whenever I reboot my PC, the keyboard shows up as "not connected" and doesn't want to connect at all. "Forget this device.", clearing my BT profile and re-pairing is the only thing that works. After I enabled debug and rebooted, the keyboard periodically sent the same error as in this issue.

BambooRaptor, 2024-06-29:
> Same here. My keyboard (running ZMK from July 2023) would not pair with my new Windows 11 laptop. I Rebuilt with 3.5 just now and it paired to the same win 11 laptop w/o any problems. I did not have to reset keyboard settings, roll back Bluetooth drivers or anything else besides update to ZMK 3.5

How did you rebuild with 3.5? Did you change something in the build.yml file? As far as I can tell, 3.5 is on the main branch now, and so should be the default choice when building, but I'm still facing this issue...

BambooRaptor, 2024-06-30:
> Since it's a long thread and this hasn't been mentioned in a while; might be worth trying [`CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y`](https://zmk.dev/docs/config/bluetooth#kconfig) if you're having issues.

It worked! Thank you!!

MatchstickWorks, 2024-07-25:
> Since it's a long thread and this hasn't been mentioned in a while; might be worth trying [`CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y`](https://zmk.dev/docs/config/bluetooth#kconfig) if you're having issues.

I tried this on my setup and it's not working. I believe I'm at the latest ZMK build (unsure how to check that against my config). I'm not using an intel driver for bluetooth, either. So I'm feeling a little stuck.

Works on my android phone, and my wife's macbook, but not her work Intel Windows PC ("Try connecting your device again"), nor my own AMD W10 PC, or my work Intel W10 PC.

Any additional thoughts?

rayanpiro, 2024-07-26:
> Since it's a long thread and this hasn't been mentioned in a while; might be worth trying [`CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y`](https://zmk.dev/docs/config/bluetooth#kconfig) if you're having issues.

Working for me too with Zephyr 3.5.0 and Corne keyboard with NRF52840 and USB 5.3 dongle on Win11.

MatchstickWorks, 2024-07-26:
> I tried this on my setup and it's not working. I believe I'm at the latest ZMK build (unsure how to check that against my config). I'm not using an intel driver for bluetooth, either. So I'm feeling [...]

Solution I had was that one of the dongles I was using was below BT4.2, so it was never going to connect. The other is that my small macropad wasn't set up to clear profiles and I didn't understand that it didn't clear profiles when you flash the firmware. So I added a setup to be able to switch and clear profiles on the board itself.

FearlessSpiff, 2024-10-11:
So cool. `CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES` also worked for me. Was a bit surprised, that I had to enter a PIN on the keyboard, but it worked!

alexsohn1126, 2024-10-26:
> Since it's a long thread and this hasn't been mentioned in a while; might be worth trying [`CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y`](https://zmk.dev/docs/config/bluetooth#kconfig) if you're having issues.

I have one PC that I dual-boot Windows and Linux on, and could not have 2 different profiles working for 2 different OS, but this fixed it. Awesome job!

FearlessSpiff, 2024-10-26:
First for me too. But now it stopped working. I went back for other things but nothing works. I read the dual boot part of the wiki https://zmk.dev/docs/troubleshooting/connection-issues#issues-with-dual-boot-setups but I don't understand it. Is it even not possible to pair on two different profiles with the same nice nano?

FearlessSpiff, 2024-10-26:
So it's not even that hard to do. Just synced my dual boot devices (Keyboard and Mouse) with the script linked in the arch wiki: https://github.com/nbanks/bluetooth-dualboot

The ZMK NiceNano was easy, as I could just copy paste the keys from the script output. For my Logitech Mouse I had do copy more things and most importantly rename the folder as the MAC address on Windows was another one than on Linux.

SamBobBarnes, 2025-01-06:
> Since it's a long thread and this hasn't been mentioned in a while; might be worth trying [`CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y`](https://zmk.dev/docs/config/bluetooth#kconfig) if you're having issues.

This worked form me! Windows 11, modified corne, tplink bluetooth dongle.

haywirephoenix, 2025-01-30:
I'm still having this issue with latest zmk. If I pair to windows, then android on profile 2, I can no longer connect back to windows without flashing a settings reset and reflash firmware.

I have tried adding all of the following to the conf:

CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y
CONFIG_ZMK_BLE_PASSKEY_ENTRY=y
CONFIG_ZMK_KSCAN_DEBOUNCE_PRESS_MS=0
CONFIG_ZMK_KSCAN_DEBOUNCE_RELEASE_MS=5
CONFIG_BT_CTLR_PHY_2M=n
CONFIG_BT_CTLR_TX_PWR_PLUS_8=y

----

nice!nanos v1
RZ616 Bluetooth Adapter
Windows 11 23H2

MatchstickWorks, 2025-01-30:
> I'm still having this issue with latest zmk. If I pair to windows, then android on profile 2, I can no longer connect back to windows without flashing a settings reset and reflash firmware.
>
> I have [...]

Do you have a key to reset a BT profile?

haywirephoenix, 2025-01-30:
> Do you have a key to reset a BT profile?

I have BT_CLR key, which I try on both halves aswell as switching back and forth with the BT_SEL keys. I could add a BT_DISC key? Since I've disabled "Allow Winedoze to put the adaptor to sleep" I can get it to respond again after hitting BT_SEL on the same profile again. It seems to work until the nanos have sat for a while, (I don't know if they have a sleep state?)

[1 more reply]