zmkfirmware/zmk #807: Feature request: Key Switch/Morph

jordykoppen, 2021-05-24:
I'd like to propose a feature that introduces support for switching a given key on the fly.

My specific use case for this is when working inside virtual machines on Windows, the GUI (or Win) keys still respond to certain Windows shortcuts like Win + L (locking the PC). If you're familiar with window/tiling mangers like i3 or BSPWM, this can be highly infuriating. 
As I haven't found a proper way of disabling this behavior inside the OS, I would love to see a feature where I can switch/morph the output of a specific key by toggling between two given keycodes.

TL;DR:
A feature which adds the possibility to switch the behavior of a certain key upon boot or by pressing a specific key. I think the functionality would be a nice addition to the already rich set of features that ZMK currently supports.

For example, a by adding a new behavior, e.g. `&sw` short for `switch`, a key can swap between two specified keys/modfiers:
```
&sw LEFT_GUI K_APP
```
The above example suggests a free format implementation where you specify which keys to swap.

**Now my only question/concern**, which I didn't think of until I wrote out all of the above is: How does ZMK know which one of the two to use? I would love to hear some suggestions, the two I came up with right now:

1. Switching between the two by holding the designated key on Reset/Power on
2. Toggling the behavior by another designated key.

Feel free to add suggestions, additions or ask for further explanation,

okke-formsma, 2021-05-24:
The easiest way is to do this at runtime.  `&sw LEFT_GUI K_APP` swaps all future uses of `LEFT_GUI` with `K_APP`. Press it again to turn it off.

So, before pressing `&sw LEFT_GUI K_APP`, `&kp LEFT_GUI` would output a LEFT_GUI keycode. After pressing the `&sw` key, it would output a `K_APP` keycode. After pressing the  `&sw LEFT_GUI K_APP` key again, the original behavior is restored.

okke-formsma, 2021-05-24:
To get to work on this, here are some pointers;

* Do you have a working local zephyr build? Make sure you can build your own keymap, and run the tests. 
* Configure your development environment so you can start a GDB debugger on the tests  (optional but very useful).
* Read behavior_key_press.c and behavior_sticky_key.c. keypress is the simplest behavior possible. It fires off a keycode_state_changed event on press, and another of those events on release. Sticky keys are more advanced, they show how to use timers (not needed here) and how to intercept keypress events (we'll need that).
* You'll have to define a new behavior. You can take https://github.com/zmkfirmware/zmk/pull/731 as example for all the things you'll need.

I'd start off with a simple version of the behavior and build from there. For example;
1. When pressing the `&sw` key, the first press outputs the first parameter, the second press outputs the other. (This is not even close to the behavior you want, but bear with me).
2. Make the behavior store the 'translation' in a global variable
3. Create a keycode_state_changed listener that logs something whenever it sees the modifier that must be translated
4. Modify the keycode_state_changed struct, removing the old modifier and adding the translation.
5. Write tests, identify bugs (what happens if you press `&sw LGUI K_APP` while a `&kp LGUI` is pressed?), fix bugs.