zmkfirmware/zmk #808: Fix warnings when enabling newlib_libc

okke-formsma, 2021-05-25:
After enabling `CONFIG_NEWLIB_LIBC`, I got the following warnings:
```
[49/272] Building C object CMakeFiles/app.dir/src/ble.c.obj
In file included from /home/okke/.local/zephyr-sdk-0.11.2/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:800,
                 from ../../src/ble.c:12:
../../src/ble.c: In function 'set_profile_address':
../../src/ble.c:118:27: warning: '%d' directive writing between 1 and 3 bytes into a region of size 2 [-Wformat-overflow=]
  118 |     sprintf(setting_name, "ble/profiles/%d", index);
      |                           ^~~~~~~~~~~~~~~~~
../../src/ble.c:118:41: note: format string is defined here
  118 |     sprintf(setting_name, "ble/profiles/%d", index);
      |                                         ^~
In file included from /home/okke/.local/zephyr-sdk-0.11.2/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:800,
                 from ../../src/ble.c:12:
../../src/ble.c:118:27: note: directive argument in the range [0, 255]
  118 |     sprintf(setting_name, "ble/profiles/%d", index);
      |                           ^~~~~~~~~~~~~~~~~
../../src/ble.c:118:5: note: '__builtin___sprintf_chk' output between 15 and 17 bytes into a destination of size 15
  118 |     sprintf(setting_name, "ble/profiles/%d", index);
      |     ^~~~~~~
[53/272] Building C object CMakeFiles/app.dir/src/split/bluetooth/central.c.obj
In file included from /home/okke/.local/zephyr-sdk-0.11.2/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/string.h:180,
                 from /home/okke/dev/zmk/zephyr/include/bluetooth/bluetooth.h:21,
                 from ../../src/split/bluetooth/central.c:9:
../../src/split/bluetooth/central.c: In function 'split_central_discovery_func':
../../src/split/bluetooth/central.c:130:9: warning: '__builtin_memcpy' forming offset [5, 17] is out of the bounds [0, 4] of object '({anonymous})' with type 'struct bt_uuid_16[1]' [-Warray-bounds]
  130 |         memcpy(&uuid, BT_UUID_GATT_CCC, sizeof(uuid));
      |         ^~~~~~
/home/okke/dev/zmk/zephyr/include/bluetooth/uuid.h:72:45: note: '({anonymous})' declared here
   72 |  ((struct bt_uuid *) ((struct bt_uuid_16[]) {BT_UUID_INIT_16(value)}))
      |                                             ^
/home/okke/dev/zmk/zephyr/include/bluetooth/uuid.h:372:2: note: in expansion of macro 'BT_UUID_DECLARE_16'
  372 |  BT_UUID_DECLARE_16(BT_UUID_GATT_CCC_VAL)
      |  ^~~~~~~~~~~~~~~~~~
../../src/split/bluetooth/central.c:130:23: note: in expansion of macro 'BT_UUID_GATT_CCC'
  130 |         memcpy(&uuid, BT_UUID_GATT_CCC, sizeof(uuid));
      |                       ^~~~~~~~~~~~~~~~
```

mcrosson, 2021-06-04:
Additional warnings pop up with display enabled

```
../../src/display/widgets/battery_status.c: In function 'set_battery_symbol':
../../src/display/widgets/battery_status.c:42:9: warning: '__builtin___strcpy_chk' writing 4 bytes into a region of size 2 overflows the destination [-Wstringop-overflow=]
   42 |         strcpy(text, LV_SYMBOL_CHARGE);
      |         ^~~~~~
../../src/display/widgets/battery_status.c:47:9: warning: '__builtin___strcat_chk' writing 4 bytes into a region of size 2 overflows the destination [-Wstringop-overflow=]
   47 |         strcat(text, LV_SYMBOL_BATTERY_FULL);
      |         ^~~~~~
../../src/display/widgets/battery_status.c:49:9: warning: '__builtin___strcat_chk' writing 4 bytes into a region of size 2 overflows the destination [-Wstringop-overflow=]
   49 |         strcat(text, LV_SYMBOL_BATTERY_3);
      |         ^~~~~~
../../src/display/widgets/battery_status.c:51:9: warning: '__builtin___strcat_chk' writing 4 bytes into a region of size 2 overflows the destination [-Wstringop-overflow=]
   51 |         strcat(text, LV_SYMBOL_BATTERY_2);
      |         ^~~~~~
../../src/display/widgets/battery_status.c:53:9: warning: '__builtin___strcat_chk' writing 4 bytes into a region of size 2 overflows the destination [-Wstringop-overflow=]
   53 |         strcat(text, LV_SYMBOL_BATTERY_1);
      |         ^~~~~~
../../src/display/widgets/battery_status.c:55:9: warning: '__builtin___strcat_chk' writing 4 bytes into a region of size 2 overflows the destination [-Wstringop-overflow=]
   55 |         strcat(text, LV_SYMBOL_BATTERY_EMPTY);
      |         ^~~~~~
[70/376] Building C object CMakeFiles/ap.../src/display/widgets/output_status.c.obj
In file included from /opt/zephyr-sdk-0.11.4/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:800,
                 from /workspaces/zmk-artsey/modules/lib/gui/lvgl/src/lv_misc/lv_printf.h:62,
                 from /workspaces/zmk-artsey/modules/lib/gui/lvgl/lvgl.h:35,
                 from ../../include/zmk/display/widgets/output_status.h:9,
                 from ../../src/display/widgets/output_status.c:12:
../../src/display/widgets/output_status.c: In function 'set_status_symbol':
/workspaces/zmk-artsey/modules/lib/gui/lvgl/src/lv_core/../lv_font/lv_symbol_def.h:69:35: warning: ' ' directive writing 4 bytes into a region of size between 0 and 2 [-Wformat-overflow=]
   69 | #define LV_SYMBOL_WIFI            "\xef\x87\xab" /*61931, 0xF1EB*/
      |                                   ^~~~~~~~~~~~~~
../../src/display/widgets/output_status.c:57:27: note: in expansion of macro 'LV_SYMBOL_WIFI'
   57 |             sprintf(text, LV_SYMBOL_WIFI "%i " LV_SYMBOL_SETTINGS, active_profile_index);
      |                           ^~~~~~~~~~~~~~
In file included from /workspaces/zmk-artsey/modules/lib/gui/lvgl/src/lv_core/../lv_font/lv_font.h:21,
                 from /workspaces/zmk-artsey/modules/lib/gui/lvgl/src/lv_core/lv_style.h:17,
                 from /workspaces/zmk-artsey/modules/lib/gui/lvgl/src/lv_core/lv_obj.h:20,
                 from /workspaces/zmk-artsey/modules/lib/gui/lvgl/lvgl.h:24,
                 from ../../include/zmk/display/widgets/output_status.h:9,
                 from ../../src/display/widgets/output_status.c:12:
/workspaces/zmk-artsey/modules/lib/gui/lvgl/src/lv_core/../lv_font/lv_symbol_def.h:30:40: note: format string is defined here
   30 | #define LV_SYMBOL_SETTINGS        "\xef\x80\x93" /*61459, 0xF013*/
In file included from /opt/zephyr-sdk-0.11.4/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:800,
                 from /workspaces/zmk-artsey/modules/lib/gui/lvgl/src/lv_misc/lv_printf.h:62,
                 from /workspaces/zmk-artsey/modules/lib/gui/lvgl/lvgl.h:35,
                 from ../../include/zmk/display/widgets/output_status.h:9,
                 from ../../src/display/widgets/output_status.c:12:
../../src/display/widgets/output_status.c:57:13: note: '__builtin___sprintf_chk' output between 9 and 11 bytes into a destination of size 6
   57 |             sprintf(text, LV_SYMBOL_WIFI "%i " LV_SYMBOL_SETTINGS, active_profile_index);
      |             ^~~~~~~
/workspaces/zmk-artsey/modules/lib/gui/lvgl/src/lv_core/../lv_font/lv_symbol_def.h:69:35: warning: ' ' directive writing 4 bytes into a region of size between 0 and 2 [-Wformat-overflow=]
   69 | #define LV_SYMBOL_WIFI            "\xef\x87\xab" /*61931, 0xF1EB*/
      |                                   ^~~~~~~~~~~~~~
../../src/display/widgets/output_status.c:54:31: note: in expansion of macro 'LV_SYMBOL_WIFI'
   54 |                 sprintf(text, LV_SYMBOL_WIFI "%i " LV_SYMBOL_CLOSE, active_profile_index);
      |                               ^~~~~~~~~~~~~~
In file included from /workspaces/zmk-artsey/modules/lib/gui/lvgl/src/lv_core/../lv_font/lv_font.h:21,
                 from /workspaces/zmk-artsey/modules/lib/gui/lvgl/src/lv_core/lv_style.h:17,
                 from /workspaces/zmk-artsey/modules/lib/gui/lvgl/src/lv_core/lv_obj.h:20,
                 from /workspaces/zmk-artsey/modules/lib/gui/lvgl/lvgl.h:24,
                 from ../../include/zmk/display/widgets/output_status.h:9,
                 from ../../src/display/widgets/output_status.c:12:
/workspaces/zmk-artsey/modules/lib/gui/lvgl/src/lv_core/../lv_font/lv_symbol_def.h:30:40: note: format string is defined here
   30 | #define LV_SYMBOL_SETTINGS        "\xef\x80\x93" /*61459, 0xF013*/
[... cut ...]
      |         ^~~~~~~
[371/376] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      284352 B       792 KB     35.06%
            SRAM:       58269 B       256 KB     22.23%
        IDT_LIST:         168 B         2 KB      8.20%
[376/376] Linking C executable zephyr/zmk.elf
Converted to uf2, output size: 568832, start address: 0x26000
Wrote 568832 bytes to /workspaces/zmk-artsey/app/build/the_paintbrush_left/zephyr/zmk.uf2
```

mcrosson, 2021-06-04:
it appears that even just increasing the size of the text fields won't produce a functional build ; something deeper is broken with display and the updated libc