zmkfirmware/zmk #815: Sticky keys inconsistent behavior in secondary layers

arhl-gh, 2021-06-01:
For example, LS(N1) gives 1 instead of ! around once every ten key presses. This behavior has been noted when the sticky key is in a secondary layer.  

Happens also for RA(code).

joelspadin, 2021-07-20:
Might be the same issue as #759

arhl-gh, 2021-07-23:
Thank you for your answer. It is a similar case indeed, the behavior seems the same although I'm using Fedora 34 instead of a remote connection.

arhl-gh, 2021-09-08:
I have found that is issue is only present if I use Wayland in Fedora. I have switched to Xorg and the erratic comportment of the sticky keys is no longer present. Please let me know if some other info could be useful to you.

caksoylar, 2022-04-03:
Closing, since this seems to be a related to the timing of keycodes with implicit modifiers similar to #759 and not related to sticky keys or non-default layers.

closed 2022-04-03 by caksoylar