zmkfirmware/zmk #828: Feature request: Multi-layer state support (including tri-layer for lower/raise/adjust)

bcat, 2021-06-09:
I was thinking of simple ZMK features I could hack on that I'd actually use, and tri-layer support (like QMK and BlueMicro firmware offer) came to the top of my list. You can fake it today by nesting momentary layer mappings, but the layer deactivation behavior on layer up isn't the same. (For example, if you have press Lower, press Raise, release Lower, you stay on the "adjust" layer until _Raise_ is released. Real tri-layer support would take you back to the "raise" layer at this point.)

I thought about implementing this feature using new layer behaviors, or about integrating with the combo system somehow, but the cleanest approach I could come up with actually involves a new node next to the keymap like so:

```
#define LOWER 1
#define RAISE 2
#define ADJUST 3

multi_layers {
  compatible = "zmk,multi-layers";
  adjust {
    if-layers = <LOWER RAISE>;
    then-layer = ADJUST;
  };
};

keymap {
  compatible = "zmk,keymap";
  default_layer {
    &kp A &kp B
    &mo LOWER &mo RAISE
  } 
  lower_layer {
    &kp C &kp D
    &trans &trans
  }
  raise_layer {
    &kp E &kp F
    &trans &trans
  }
  adjust_layer {
    &kp G &kp H
    &trans &trans
  }
};
```

I wrote a [first draft implementation](https://gist.github.com/bcat/647938081fa90304197af8ea2e1cd51d) (that I've not tested **at all**), and I'd like to get some early feedback in two areas. (Of course, it's also possible this approach is wrong on a more fundamental level. That feedback is welcome as well. 😄)

First, should multi-layer configs be applied recursively? In other words, should activating the then-layer of one config be able to trigger the if-layer of another config?

Advantages to recursive activation:

* Simpler implementation (this is how the event model works by default)
* More flexible for users

Disadvantages to recursive activation:

* Possible for a circular reference between multi-layer configs to cause permanent layer activation, since two layers could reactivate one another immediately after deactivating.

Second, should multi-layer (de)activation update the existing layer state bitset, or should there be a new bitset for multi-layer activations that gets OR'd with the existing layer state bitset to compute the "effective layers state"?

Advantages to updating the existing layers state:

* Simpler implementation (no changes needed to the core layers impl for this feature)

Disadvantages to updating the existing layers state:

* Interactions with existing layer activation mechanisms may be confusing, e.g., if a layer is simultaneously activated via both a `&mo` behavior and a multi-layer config, ideally it would be deactivated only after _both_ activation conditions stop triggering, but if the two approaches share a layer bitset, the layer will become inactive earlier, when _the first_ activation condition stops.

bcat, 2021-06-09:
Regarding OR'ing multiple layer state bitsets together to compute the effective layer state, I feel like my natural inclination would be allowing code to register a new layer state bitset, so that e.g. `keymap.c` could register one for the layer behaviors, `multi_layer.c` could register one for the multi-layer configs, and if users wanted to even more customer layer logic, they could register their own bitset in a C file in their ZMK config repo. (Assuming that's a thing you can do today; I don't actually know if it's possible to put custom source code into a config repo.) But this may be a bit overdesigned... 😄

bcat, 2021-06-09:
I guess another alternative solution to the "same layer activated from multiple places simultaneously" problem would be to store an _activation count_ for each layer instead of just an _is active_ bit. But in addition to maybe being worse from a storage perspective, I always find these type of ref-counting style approaches to be hard to reason about the correctness of when reading the code.

bcat, 2021-06-10:
Also, now that I think about it, something like "conditional layers" would probably be a clearer name. But it's tricky because the community calls the Lower + Raise = Adjust version "tri layer" and deviating too much from that name might else be confusing. IDK, naming is hard....

bcat, 2021-06-14:
Thinking about this a little bit more:

1. I really like the idea of calling this feature "conditional layers" or "layer combos" or something, and simply documenting that you can achieve "tri layer" support using it. The more I look at the time "multi layer" the less I like it. :)

2. I wonder if the "then-layer" attribute should just specify bindings like the combos feature uses. Instead of `then-layer = ADJUST`, you'd write `binds = <&mo ADJUST>`, for instance. It's more flexible and sidesteps the question about one vs. many active layer sets. (But also it may be overkill.... I can't imagine ever using a behavior other than `&mo` here.)

WDYT?

bcat, 2021-10-01:
Getting back to this, I am renaming the feature to "conditional layers" since others seemed to think that name was clearer and I agree.

I'm probably punting on the more flexible binding support; we'll see.

evantravers, 2021-10-20:
Thanks @petejohanson and @bcat. I think that I'll be ok with the combo workaround for now… my "adjust" layer is pretty intentional, reset, volume, etc. I don't really flow into it from one layer to another.

PR #830 (2021-11-15): feat(conditional-layers): Implement generalized version of tri-layer support for "adjust" layer
I missed tri-layer support from QMK, so I implemented a ZMK version. It seemed most logical to express as an N-to-1 mapping of `if-layers` to activate a single `then-layer`, so I called it "conditional layers".

As discussed in the attached issue, I considered calling this "layer combos" and replacing the `then-layer` property with a `bindings` property that could trigger an `&mo` behavior (or any other behavior as well), but in practice I didn't see any reason to generalize that much.

Fixes #828.
 [... cut]
closed 2021-11-15 by petejohanson