zmkfirmware/zmk #829: Cannot roll sticky keys

dxmh, 2021-06-09:
It’s not possible to do a combination of sticky modifiers unless each one is pressed _and released_ before the next. 

Example: Rolling `shift`+`ctrl`+`left` 

1. Press `&sk LSHIFT` 
2. Press `&sk LCTRL` 
3. Release `&sk LSHIFT` 
4. Release `&sk LCTRL` 
5. Press and release `&kp LEFT` 

Expected result:

1. Shift down
2. Control down
3. Left down
4. Left up
5. Shift up
6. Control up

Actual result

1. Shift down
2. Control down
3. **Shift up**
4. Left down
5. Left up
6. Control up

dxmh, 2021-06-10:
I think this might be related to (or even a duplicate of) #787:

> Modified keycodes or combined modifiers can't be used as sticky keys right now. They are released too early, because the sticky key doesn't recognise it's own keypress.
> 
> This will be fixed once #670 lands.

okke-formsma, 2021-06-13:
The use case where you'd want to trigger multiple hold-taps, on a homerow-holdtaps-layer for example, was not considered when building this behavior.

Sticky keys can't be rolled with other keys either; when you press another key while the sticky key is pressed, the sticky key behaves as a normal key.

Maybe we should change the behavior of sticky keys based on whether the pressed key is a modifier or not. I have to medidate a bit on this.

peauters, 2021-08-09:
Could you not have it look at whether the key pressed is another sticky key instead of modifier?

daliusd, 2021-09-11:
I am using Callum one-shot keys and layers (my improvement, details here https://blog.ffff.lt/posts/callum-layers/) and there is function `is_oneshot_ignored_key` with keys that do not trigger one-shot/sticky keys to be released. I personally added all sticky keys (both mods and layers) in it. Therefore it is possible that this would be sane default, but mods alone are clearly not enough.

jimmerricks, 2021-11-26:
I wrote a couple of behaviors that put sticky keys and layers inside of a hold tap. It allows for rolling.

```
/ {
    behaviors {
        nk: not_so_sticky_key {
            compatible = "zmk,behavior-hold-tap";
            label = "not so sticky key";
            #binding-cells = <2>;
            tapping_term_ms = <200>;
            flavor = "tap-preferred";
            bindings = <&kp>, <&sk>;
        };
        nl: not_so_sticky_layer {
            compatible = "zmk,behavior-hold-tap";
            label = "not so sticky layer";
            #binding-cells = <2>;
            tapping_term_ms = <200>;
            flavor = "tap-preferred";
            bindings = <&mo>, <&sl>;
        };
    };
};
```

okke-formsma, 2022-01-28:
I think this should solve the issue

https://github.com/okke-formsma/zmk/tree/sticky-hold

kkga, 2022-01-28:
@okke-formsma this looks great, thank you! i've had issues with the custom behavior solutions proposed in this thread, so having this built-in would be amazing.

jcmkk3, 2022-01-28:
> I think this should solve the issue
> 
> https://github.com/okke-formsma/zmk/tree/sticky-hold

Using this branch right now and so far it is working perfectly!

peauters, 2022-01-29:
Can confirm its working for me as well.

kkga, 2022-01-29:
works great for me too 👍🏼

kkga, 2022-02-06:
@okke-formsma I've been driving your for a few days and it's been working great -- thanks for doing this 👍🏼
Wondering if you're planning to PR this upstream so that we can track and switch back to upstream once it's merged :)

commit cfd0d3d8 (2022-02-07): Behaviors: Add 'ignore-modifiers' option to sticky keys

To combine multiple sticky modifiers, the sticky keys must ignore
other (sticky) modifier keypresses.

This behavior is important for "callum-style mods", where all modifiers
are sticky mods.

Fixes #829
closed 2022-02-07 by petejohanson