zmkfirmware/zmk #834: Feature request: Sticky key “cancel” key

dxmh, 2021-06-11:
### Current behaviour: 

- Sticky keys are cancelled after a timeout

### Desired behaviour:

- Sticky keys are cancelled after a timeout, OR
- Sticky keys are cancelled with a key

### Examples:

With `&kp ESCAPE` configured as cancel key:

- Tap `&sk LSHIFT`, `&kp A` = A (uppercase)
- Tap `&sk LSHIFT`, `&kp ESC`, `&kp A` = a (lowercase)

With `&mo NAV` configured as cancel key:

- Tap `&sk LSHIFT`, `&mo NAV`, `&kp A` = a (lowercase)

dxmh, 2021-09-10:
Another idea... In https://github.com/zmkfirmware/zmk/issues/829#issuecomment-895202987 this was suggested:

> Could you not have it look at whether the key pressed is another sticky key instead of modifier?

This made me think - when checking if the key being pressed is also sticky, or a modifier, or whatever, it could also check if the key being pressed is itself and then cancel/unstick if so. 

This would allow for an experience very similar to Callum Oakley's popular one-shot key implementation: https://github.com/callum-oakley/qmk_firmware/blob/master/users/callum/readme.md (documentation link).

daliusd, 2021-09-11:
I have improved Callum one-shot key implementation with layers (details here https://blog.ffff.lt/posts/callum-layers/) and found out that potentially I need different cancel keys for one-shot modifiers and one-shot layers. Just to make everything more complicated :smile:

ldebritto, 2023-01-27:
Working around this I've faked a cancel function by creating a macro to trigger a F22 tap when hitting any of the thumb layer switches:

```
    macros {
        la_nav: cancel_queued_os_mods_and_toggle_nav {
            label = "LA_NAV";
            compatible = "zmk,behavior-macro";
            #binding-cells = <0>;
            bindings
                = <&macro_press &mo NAV>
                , <&macro_tap &kp F22>
                , <&macro_pause_for_release>
                , <&macro_release &mo NAV>
                ;
        };
        la_sym: cancel_queued_os_mods_and_toggle_sym {
            label = "LA_SYM";
            compatible = "zmk,behavior-macro";
            #binding-cells = <0>;
            bindings
                = <&macro_press &mo SYM>
                , <&macro_tap &kp F22>
                , <&macro_pause_for_release>
                , <&macro_release &mo SYM>
                ;
        };
```