zmkfirmware/zmk #837: Bluetooth pairing rejected by ZMK board

CorruptMilkshake, 2021-06-12:
I'm having an issue where the board (nice!nano microcontroller with Lily58 PCB) does not connect via bluetooth. The error message on Windows is "Try connecting your device again" and error message on Android is "Couldn't pair with Lily58 Left because of an incorrect PIN or passkey." USB connection is functional, as is communication between halves.

The firmware is the default nice!nano lily58 configuration, following instructions on the ZMK website to the letter, with the only change being enabling logging. What seems to be the relevant section of the log is below.

`[00:02:00.329,833] [0m<dbg> zmk.bvd_sample_fetch: ADC raw 4088 ~ 2994 mV => 4200 mV[0m`
`[00:02:00.329,833] [0m<dbg> zmk.bvd_sample_fetch: Percent: 100[0m`
`[00:02:52.548,217] [0m<dbg> zmk.split_central_connected: Connected: 43:89:b6:bf:4c:6b (random)[0m`
`[00:02:52.548,217] [0m<dbg> zmk.split_central_process_connection: Current security for connection: 1[0m`
`[00:02:52.549,560] [0m<dbg> zmk.split_central_process_connection: New connection params: Interval: 36, Latency: 0, PHY: 1[0m`
`[00:02:52.549,621] [0m<dbg> zmk.connected: Connected thread: 0x20002a88[0m`
`[00:02:52.549,652] [0m<dbg> zmk.connected: Connected 43:89:b6:bf:4c:6b (random)[0m`
`[00:02:52.549,652] [0m<dbg> zmk.update_advertising: advertising from 0 to 2[0m`
`[00:02:52.650,482] [0m<dbg> zmk.auth_pairing_accept: role 1, open? no[0m`
`[00:02:52.650,482] [1;33m<wrn> zmk: Rejecting pairing request to taken profile 0[0m`
`[00:02:52.651,306] [1;31m<err> zmk: Security failed: 43:89:b6:bf:4c:6b (random) level 1 err 8[0m`
`[00:02:55.710,510] [0m<dbg> zmk.split_central_disconnected: Disconnected: 43:89:b6:bf:4c:6b (random) (reason 19)[0m`
`[00:02:55.710,601] [0m<dbg> zmk.disconnected: Disconnected from 43:89:b6:bf:4c:6b (random) (reason 0x13)[0m`
`[00:02:55.710,601] [0m<dbg> zmk.update_advertising: advertising from 2 to 2[0m`
`[00:03:00.329,833] [0m<dbg> zmk.bvd_sample_fetch: ADC raw 4090 ~ 2995 mV => 4201 mV[0m`
`[00:03:00.329,833] [0m<dbg> zmk.bvd_sample_fetch: Percent: 100[0m`

Any help in fixing this would be very much appreciated. Thanks

Nicell, 2021-06-12:
@CorruptMilkshake Are you managing your bluetooth profiles? Please make sure you're using an open profile when trying to connect to a new device. I believe this is failing due to a prior connection to another device. Take a look at our docs here: https://zmk.dev/docs/behaviors/bluetooth#bluetooth-pairing-and-profiles

When connecting a device, you must have a cleared profile that's ready to accept a new bond.

CorruptMilkshake, 2021-06-12:
@Nicell Huh, wow. I don't remember successfully managing to pair it before, but I must have done because clearing the profiles it all works.

Thank you very much.

closed 2021-06-12 by CorruptMilkshake