zmkfirmware/zmk #841: BUG: USB Power detected as a host

hahuang65, 2021-06-15:
Having a USB-power cable drive the keyboard (not attached to a laptop, cable without data) still has the firmware thinking it's attached to a host, which sends data over USB, instead of BLE.

If the power source isn't detected to be an actual host, it would be nice to automatically switch over to BLE transmission.

i7d3v3l0p3r, 2021-06-21:
Temporary solution: use `&out OUT_BLE` binding (https://zmk.dev/docs/behaviors/outputs)

piit79, 2024-04-11:
Oh, I just got caught out by this issue. I see it's been going on for *a while*...

Is it realistically possible to get it fixed? QMK uses USB activity for sidedness detection, would it be possible to somehow monitor USB activity to decide if there's a host connected to USB?

angweekiat, 2024-10-25:
Hi @hahuang65 @piit79 @i7d3v3l0p3r , this may be premature, but I'm happy to say that the fix PR for this issue is merged into `main` above. As mentioned in the PR, existing documentations (https://zmk.dev/docs/keymaps/behaviors/outputs) will still mention this issue, until we have more real-world testing with the changes.

hahuang65, 2024-10-25:
Thanks for all the hard work, to ALL the ZMK contributors!

petejohanson, 2024-10-25:
> Thanks for all the hard work, to ALL the ZMK contributors!

Can you please confirm if current ZMK `main` addresses the issues you were encountering?

piit79, 2024-10-25:
I just tested with current `main` and I can confirm the board is no longer switching to USB mode when powered from a charger/power bank. Thanks @angweekiat, great job!