zmkfirmware/zmk #852: Utilize Vial for Realtime Keymap Updating

luigi311, 2021-06-22:
For QMK the default Realtime Keymap Updating tool is VIA, I am proposing utilizing Vial as the tool for remaping.

Vial has the benefit of being open source so if the developer ever abandons it, it can still be updated and maintained by a third party. They also have some security benefits to prevent the keys from being switched to something else without the users consent. Keyboards also do not have to be maintained on the Vial side of things as the json will live inside the firmware and retrieved dynamically so keyboard support just needs to be handled on a single repo.

More information is here
https://get.vial.today/
https://github.com/vial-kb

lucastercas, 2022-12-11:
So, is there any tool that I can use to update the keymaps realtime? Without having to edit the `.keymap` file and reflashing?

caksoylar, 2022-12-11:
There isn't one at the moment, no. There is no support on the ZMK backend for such an application yet.

lublak, 2023-07-17:
Just my idea.
I currently use an iPad. One dream would be to create a keyboard that can change the keyboard Lay-out in real time over Bluetooth.
So a backend that support Bluetooth keymap remapping would be awesome. So after some time there can also be a iPad app to configure the keyboard. (Flutter would be a nice framework for this, it works on android, iPad, windows, Linux and Mac)

myst729, 2024-03-08:
Given that ZMK supports UF2 firmware update, is it possible to just drag a keymap file (YAML, JSON or whatever, even plain TXT) under its DFU mode?

Nick-Munnich, 2024-08-07:
Closing as ZMK is planned and will soon be using ZMK studio to attain the same effect.

closed 2024-08-07 by Nick-Munnich

cmandlbaur, 2024-08-26:
Ok, but how soon is soon™️?

Nick-Munnich, 2024-08-26:
> Ok, but how soon is soon™️?

Currently in alpha, can be tested. See the Discord for more details, also see https://main.alpha.zmk.studio/ and #2268