zmkfirmware/zmk #874: trouble with bluemicro840_v1 split

sanderboer, 2021-07-18:
Hi,

I have built a wireless split with two bluemicro840's installed and it is in working order with a BlueMicro_BLE fw, thanks to the tireless help of JP Constantineau on the bluemicro discord.

The kb only has 34 keys, it relies heavily on layer tap, homerow modifiers and combos. I made a non-split version of ZMK firmware  to test these requirements and I found them to be working flawlessly and to be very responsive.

However, when I flash a split version of the ZMK firmware I have trouble connecting the master (left side) to the computer and I have yet to succeed in having the left side talk to the right side. I have flashed the bluemicro840_v1-settings_reset-zmk a number of times to no avail. When connecting either half to the computer with usb I can see the layers, combos and layer-tap behavior is all working perfectly. 

Somehow I cannot see this behavior separate from the trouble I am having connecting the master to the computer, but maybe it is unrelated. In any case, when trying to pair, bluetoothctl does not show the pairing connection request. Similarly, on android the connection is failing with the error that the pin was incorrect.

If anyone is interested, my fork with the board code is here:
https://github.com/sanderboer/zmk/tree/lil_chonky/app/boards/shields/lil_chonky_bois

I would appreciate some information or tips on steps I can undertake to further investigate this issue, as I am very enthusiastic about the responsiveness of my initial firmware tests.

MvEerd, 2021-07-18:
> When connecting either half to the computer with usb I can see the layers, combos and layer-tap behavior is all working perfectly.

Are you saying that the right-hand (peripheral) side is working over USB when plugged in?
This would indicate that it has been flashed with a firmware that has a `central` role, since the peripheral half should not work over USB normally. It only works by connecting through the left side.
Makes sure your uf2 files are for the correct side of the keyboard of each half

If they only work when plugged in to USB for power, but are working they are likely connected wirelessly but not powered without USB, check that the battery voltage is correct.

> Similarly, on android the connection is failing with the error that the pin was incorrect.

Are you selecting a clear [bluetooth profile](https://zmk.dev/docs/behaviors/bluetooth) for each device you're pairing with? 

Your configuration looks fine at first glance, the left is being set as the CENTRAL device. You could explicitly

sanderboer, 2021-07-19:
Hi Michael,

Thank you for your kind reply, it does further my understanding of ZMK and lessen my fear of the voodoo that is bluetooth.

However, I am saddened to have to report that this issue is an outstanding example of [PEBKAC](https://www.urbandictionary.com/define.php?term=pebkac) as I overlooked a typo in my build script, which resulted in me flashing the wrong fw to the left side. 

This issue cannot be closed fast enough as it makes me cringe thinking about the time you graciously afforded me in replying.

closed 2021-07-19 by sanderboer