zmkfirmware/zmk #881: Mod-tap: allow sending modifier immediately on key down

lyallcooper, 2021-07-27:
As far as I can tell from the various configuration options around hold-tap, there is no way to replicate this desired behavior:

**key**: `&mt LGUI ESC`

Within tapping term:
key down: `LGUI` down event sent
key up: `LGUI` up event sent, then `ESC` down + up sent

When held longer than tapping term:
key down: `LGUI` down event sent
key up: `LGUI` up event sent
***
### Why this is useful:
This allows someone to e.g. `GUI` + click immediately after pressing the key, instead of having to wait for the tapping term to expire first.

caksoylar, 2023-10-14:
#1811 would resolve this.

akira-toriyama, 2023-10-14:
@caksoylar Thank you

caksoylar, 2025-02-01:
Closing since #1811 handled it.

closed 2025-02-01 by caksoylar