zmkfirmware/zmk #891: References to Pro Micros in the docs are confusing for new users.

JakeHL, 2021-08-02:
For context: I've recently discovered ZMK after being a long time QMK user and have been looking for an appropriate board to use with ZMK that's in stock and reasonably priced in the UK.

Upon reading through the documentation there are several references to Pro Micros that lead me to believe that they are compatible boards (and thus, similar boards with ATMega based chips). One such reference can be found [here](https://zmk.dev/docs/development/new-shield#shield-overlays).

@petejohanson clarified in discord that it is all about the footprint/pinouts which the compatible boards conform to in order to be "drop in replacements". This makes sense, however it's not clear for a new user reading only the docs. 

A proposed solution would be to update all graphics/text references to a similar compatible board or clarify in the supported section that the Pro Micro (and other ATMega hardware) is not supported.

rhhub, 2022-03-29:
This happened to me as well. On this page https://zmk.dev/docs/hardware The heading structure is as follows: 

Supported Hardware/Keyboards/Composite Boards/Pro Micro Keyboards: 

`The SparkFun Pro Micro grew popular as a low cost ATmega32U4 board with sufficient GPIO and peripherals to work for many keyboard needs. Since the original Pro Micro, many pin compatible boards have appeared, with various changes or improvements, such as the Elite-C w/ USB-C, nice!nano with nRF52840 wireless. `

It's ambiguous. It's a history lesson inside a list of compatible boards.

rhhub, 2022-03-29:
I tried to make a commit. The following page https://github.com/zmkfirmware/zmk/edit/main/docs/docs/hardware.mdx references "@site/src/data/hardware-metadata.json" but it is ignored at https://github.com/zmkfirmware/zmk/blob/main/docs/src/data/.gitignore I have no idea how to edit it.

joelspadin, 2022-03-29:
That file is generated at build time by collecting all the *. zmk.yml files in the repo.

An easy way to find the source of that text is to just search the repo for some phrase in it. The file you're looking for is https://github.com/zmkfirmware/zmk/blob/main/app/boards/interconnects/pro_micro/pro_micro.zmk.yml

rhhub, 2022-03-29:
Thanks for the sanity check. I searched for the heading "Pro Micro Keyboards" and nothing showed up.

rhhub, 2022-03-29:
Relevant pull requests:
[Emphasize AMR processor incompatibility with ZMK](https://github.com/zmkfirmware/zmk/pull/1199)
[Cleared up ambiguity on supported hardware page](https://github.com/zmkfirmware/zmk/pull/1200)

caksoylar, 2022-04-03:
Thanks @rhhub for the docs improvements! I checked all instances of "Pro Micro" in the docs and now they all seem to emphasize "compatible" or "format". Is this safe to close or do you think there are any specific parts left that are confusing?

rhhub, 2022-04-03:
The extent of my review was on the supported hardware page and I'm fine with closing because I think that is where the information is most pertinent. The use of pro micro graphics on other pages, such as the one on the OP's provided link, is fine by me.

closed 2023-11-05 by caksoylar