zmkfirmware/zmk #898: Copying the resulting UF2 to Particle Xenon (nrf52840) does not work

dzervas, 2021-08-06:
I'm trying to flash ZMK to the Particle Xenon (nrf52840, board defined in https://github.com/zephyrproject-rtos/zephyr/tree/main/boards/arm/particle_xenon) and I've made various changes in the overlay so that it's able to support zmk (https://github.com/dzervas/zmk-config/tree/main/config/boards/shields/lab68).

It compiles as expected and results in a `.bin` which I then transform to `.uf2` here https://github.com/dzervas/zmk-config/blob/main/.github/workflows/build.yml#L44

The problem is that when I copy the resulting .uf2 to the particle xenon (I've flashed adafruit's bootloader and it works with circuitpython for example) it just resets then acts like it has nothing to run. I speculated problems with the `flash0` partitions and I redefined them: https://github.com/dzervas/zmk-config/blob/main/config/boards/shields/lab68/lab68.overlay#L90-L135 but without any luck.

Any ideas? How can I debug that? I have a debugger so maybe a pointer towards that? Should I flash the bin via openocd or adafruit-nrfutil? can I get any logs from the UF2 bootloader?

dzervas, 2021-08-06:
It seems that the firmware overwrites the softdevice for some reason... Investigating the flash partitions layouts...

dzervas, 2021-08-07:
ok, I worked around the problem but here's the rundown (with the amazing and instant help of @petejohanson ):
- have "normal" adafruit bootloader with SoftDevice and trying to flash without altering the DTS of Xenon (defined in upstream Zephyr). Of course it didn't work as Zephyr had no idea where all the adafruit stuff was. I tried playing around with the nice_nano offsets (which uses SoftDevice) without any luck
- have a "nosd" bootloader and try to play around with the offsets based on https://github.com/petejohanson/zaphod-config/blob/main/boards/arm/zaphod/zaphod.dts which does exactly that. The idea behind ditching SoftDevice is that it's completely useless with ZMK/Zephyr - Zephyr has its own open source BT stack. The code offset should be `0x1000` but I had no luck
- I was tired so I just disabled `CONFIG_USE_DT_CODE_PARTITION` and flashed the resulting hex at 0x0. That means that I ditched the bootloader. I have a debugger at hand and can flash whenever I like. Don't do this if you don't have a debugger.

Command to flash with openocd (and particle debugger as my target was particle xenon):
`openocd -f interface/cmsis-dap.cfg -f nrf52-particle.cfg -c "adapter_khz 1000" -c "init" -c "reset init" -c "halt" -c "program build/zephyr/zmk.hex 0x0 verify reset" -c "exit"`

`nrf52-particle.cfg`
```
# Nordic nRF52 series: ARM Cortex-M4 @ 64 MHz
#

source [find target/swj-dp.tcl]

if { [info exists CHIPNAME] } {
       set _CHIPNAME $CHIPNAME
} else {
       set _CHIPNAME nrf52
}

# Work-area is a space in RAM used for flash programming
# By default use 16kB
if { [info exists WORKAREASIZE] } {
   set _WORKAREASIZE $WORKAREASIZE
} else {
   set _WORKAREASIZE 0x4000
}

if { [info exists CPUTAPID] } {
       set _CPUTAPID $CPUTAPID
} else {
       set _CPUTAPID 0x2ba01477
}

swj_newdap $_CHIPNAME cpu -expected-id $_CPUTAPID
dap create $_CHIPNAME.dap -chain-position $_CHIPNAME.cpu

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME cortex_m -dap $_CHIPNAME.dap

adapter_khz 1000

$_TARGETNAME configure -work-area-phys 0x20000000 -work-area-size $_WORKAREASIZE -work-area-backup 0

if { ![using_hla] } {
       cortex_m reset_config sysresetreq
}

flash bank $_CHIPNAME.flash nrf5 0x00000000 0 1 1 $_TARGETNAME
flash bank $_CHIPNAME.uicr nrf5 0x10001000 0 1 1 $_TARGETNAME

#
# Added by Particle below
#
if { [info exists ENDIAN] } {
   set _ENDIAN $ENDIAN
} else {
   set _ENDIAN little
}

proc bld_addr {bld_addr} {
    mww 0x4001e504 0x00000002;
    mww 0x4001e514 0x00000001; sleep 1000;
    mww 0x4001e504 0x00000001;
    mww 0x10001014 $bld_addr; sleep 1000; mdw 0x10001014
    mww 0x4001e504 0x00000000;

    reset run
}

$_TARGETNAME configure -event gdb-attach {
   halt
}

$_TARGETNAME configure -event gdb-attach {
   reset init
}
```

closed 2021-08-07 by dzervas

DuMaM, 2021-11-04:
@dzervas  this is how i handled particle xenon 
https://github.com/DuMaM/zmk-config/blob/master/config/boards/arm/particle_xenon/particle_xenon.overlay

dzervas, 2021-11-04:
Thank you @DuMaM! I was fed up and finally went the mcuboot way and it's working great https://github.com/dzervas/zmk-config/blob/main/config/boards/shields/lab68/lab68_mcuboot.overlay

DuMaM, 2021-11-08:
So you are suggesting that issue maybe connected with boot loaders?