zmkfirmware/zmk #905: Overlapping combos need to have the same timings (bug)

mcrosson, 2021-08-10:
While working on ARTSEY we tuned a number of combo's timings (overlapping combos) to have different timeouts and now the typed keys can be intermixed/missed/incorrectly typed.

After discussing with @okke-formsma it looks like there is a bug w/ the combo handling code if the overlapping combos have different timouts set.

Code : https://github.com/artseyio/zmk-artsey/blob/main/config/artsey.dtsi#L113
Discord discussion: https://discord.com/channels/771908552818294784/809504011199578173/874543633330749440

urob, 2022-08-01:
@petejohanson, @okke-formsma and @andrewjrae commited a sequence of combo fixes (collected in https://github.com/zmkfirmware/zmk/pull/1411) that have solved the issue for me. Could that also be the fix for the ARTSEY?

petejohanson, 2022-08-04:
@mcrosson This should now be fixed, as part of the work in #1411 where part of the fix included a fix to avoid copying the same lowest timeout value whenever re-evaluating remaining combos. Marking closed, please re-open if you still encounter this in ARTSEY.

closed 2022-08-04 by petejohanson