zmkfirmware/zmk #913: USB HID does not work on macOS FileVault login screen

xudongzheng, 2021-08-15:
If FileVault (full disk encryption) is turned on in macOS, ZMK over USB currently cannot be used on the boot login screen before the disk is decrypted. For comparison, I have a old QMK board that does work in this scenario.

Not sure what's going on and haven't had the time to dig into this but a few possibilities off the top of my head:

- Reliance on boot protocol
- Incompatible report descriptor
- Some problem in the USB/HID protocol

rstacruz, 2021-11-17:
I can confirm that my ZMK keyboard doesn't work on the FileVault screen either (Microdox, nice!nano v2, zmk 2.5). However, QMK keyboards work just fine.

rbitanga, 2022-02-21:
I have this working with Monterey but don't have older versions to test with. It will most likely at the very least work for all Macs released within the past decade since the HID spec hasn't been updated since 2001 and the official spec appears to be exported from MS Word 2000.

xudongzheng, 2023-06-09:
It seems that the FileVault login on my M1 MacBook running macOS 13 supports both Bluetooth keyboards and USB HID report protocol (could be new for M1 Macs, could be new in macOS 13, not sure).

amadeus, 2023-07-25:
Definitely noticed this issue on my two macs -- an Intel iMac and Macbook Pro M1, both with macOS 13 (Ventura). At the very least it would be great if they could work over USB. My QMK keyboards have no problem with this.

xudongzheng, 2023-11-18:
This is closed by #1793.

closed 2023-11-18 by xudongzheng